meshtastic/firmware #496: Update build system to support SPIFFS on Esp32

mc-hamster, 2020-10-20:
Update build system to support SPIFFS on esp32.

Also need to test how this will work with the OTA update system.

May need to investigate merging the two .bin files that will be generated.

mc-hamster, 2020-10-20:
It's very likely that updating the spiffs from an image will delete db.proto (I think we use that to persist device settings). That will need to be considered, especially for the OTA update process. Maybe load db.proto into memory prior to OTA and then write it back onto the firmware. May also be possible to further portion the flash memory and put db.proto out of reach of the flash update system.

geeksville, 2020-11-15:
yep - I'm working on this (finally).  I'll resave node.db (and the preferences) after updating the spiffs over BLE.

notes from the console from building the spiffs image:
```
Building SPIFFS image from 'data' directory to .pio/build/tbeam/spiffs.bin
/static/basic.js
/static/index.html
/static/meshtastic.js.gz
Looking for upload port...

Warning! Please install `99-platformio-udev.rules`. 
More details: https://docs.platformio.org/page/faq.html#platformio-udev-rules

Auto-detected: /dev/ttyUSB0
Uploading .pio/build/tbeam/spiffs.bin
esptool.py v2.6
Serial port /dev/ttyUSB0
Connecting....
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
MAC: 24:62:ab:f8:40:98
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 458752 bytes to 27837...

Writing at 0x00390000... (50 %)
Writing at 0x00394000... (100 %)
Wrote 458752 bytes (27837 compressed) at 0x00390000 in 0.5 seconds (effective 8052.5 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```

commit d7d13d63 (2020-11-28): clean up filesystem goo, add spiffs to install scripts, fix #496
@mc-hamster seems to work pretty good for me, so I'll send a PR to you
for the dev-http branch.

I'll push out an android alpha build later today (once the build is
complete).  Once this new device load is out in the field _future_
device builds will support updating spiffs from android. (i.e. device
loads older than 1.1.9 must be updated to 1.1.9 or later before spiffs
support is implemented on the device side - so some users might need to
update twice before the new spiffs contents will appear on their device)
closed 2020-11-28 by mc-hamster