meshtastic/firmware #497: RFC - dual configuration to support disconnected devices

a-f-G-U-C, 2020-10-20:
### Background

In the landscape of personal mesh messengers, Meshtastic has the fairly unique property that it can either be paired to a smartdevice and become effectively a peripheral of that smartdevice, or operate completely stand-alone on its own set of rules. From a use-case angle, these two modes are so fundamentally different that they might as well be two different types of device. This is good. 

### Problem

Sometimes, a device can change from managed to stand-alone mode and back without the user’s knowledge or intent. For example, a device that’s accidentally dropped behind during a hiking trip. Or a team using a bag of identical devices on a random pick basis. 
In these cases, the personalized settings set by the smartdevice app while the device is in managed mode aren’t necessarily equally useful once the device becomes disconnected. However in the cases of accidental or hasty disconnection, it’s usually impossible to change or reset the device configuration to a more useful/desirable set of settings - because the device is already out of reach. 

### Discussion

Wouldn’t it be great if, when the Meshtastic device gets disconnected from the managing smartdevice, it would automatically reconfigure itself to a fallback set of settings pre-programmed by the owner (while the device was still in reach)? It would solve the older complaint about unwanted persistence of settings, would simplify the recovery of lost devices and might even *improve network security* (more on this later).

It would be useful, from this perspective, for the device firmware to support **two sets of settings** switched automatically: one for managed mode, second one for disconnected/stand-alone mode. 

The connect-disconnect events are already intercepted by the firmware for power management purposes, so hooking another action at these points should be fairly uncomplicated. 

Obviously this change would affect all Meshtastic system elements: device, client, API etc, and it’s possible that its complexity may outweigh the benefits. I am opening this issue to hopefully get some community feedback on the idea.

Thank you.

**Related discussion:** https://github.com/paulmandal/atak-forwarder/issues/53

cyclomies, 2020-10-21:
I do like your concept of managed and unmanaged devices! 

Your request can be answered by either modifying the current code, or by developing the current network stack.

By modifying the current code, there could be double settings for managed and unmanaged situations. The status could be chosen depending of the Bluetooth connectivity, with a sufficient delay function. I believe this modification might be relatively easy for the skilled people coding this project, if they have the time for this.

In the other hand, by developing the Meshtastic mesh network stack (and the firmware), by dividing user related settings and parameters, from mesh protocol related settings, could achieve the same. This would require, according to my understanding, considerably more work to be done. However, dividing those would be beneficial for the next steps in developing the mesh protocol further.

The networking module (physical execution and logical transfer protocols) would serve only the networking features (BT, WLAN, automatic reconstruction of the mesh, sending, forwarding, routing and receiving messages) between nodes and to users mobile devices. Therefore, the base firmware could be smaller, and no user related information would be stored directly on the node (additional security).

The module handling user related information could be “flashed” to the node. This would let the node act as a “user device”, being able to decrypt and encrypt messages for the user to be shown on the screen. To be able to do this, the PSK, and other user related setting should be stored directly on the device (current status). 

Regarding to your RFC, the divided model could serve your need for managed and unmanaged modes, with added security (as unmanaged devices could still serve the mesh, without user related information). The device could delete user related settings, from the node, within X min, after the BT connection is lost (depending on chosen parameters). The node could recover user related settings from the user’s mobile device, once the node is reconnected to the mobile device with BT.

Unencrypted (i. e. emergency messages), or encrypted messages with a different PSK, could still be shown on unmanaged devices, if choosen. Nodes could also be permitted to send their GPS location, if chosen.

mc-hamster, 2020-12-15:
This would be possible with the new plug-in system.