meshtastic/firmware #506: v1.1.7 - possible GPS related regression on tbeam-Modle

a-f-G-U-C, 2020-10-30:
**Description**
After upgrading device firmware from v1.1.6 to v1.1.7 on a tbeam1.1-M8N-SX1262 ("Modle"), the standalone location broadcast feature appears to have stopped working. Downgrading to 1.1.6 fixes the problem.

**To Reproduce**
Steps to reproduce the behavior:
0. connect tbeam board to USB
1. upload firmware-1.1.6.bin
2. perform initialization sequence:
  meshtastic --set factory_reset 1
  meshtastic --set gps_operation 2
  meshtastic --set location_share 1
  meshtastic --set position_broadcast_secs 30
  meshtastic --seturl "http://www.meshtastic.org/..."
3. reset board and allow to sync GPS
4. from a second device, watch the packets (for example, using the Debug Panel in the app) - "position" messages should contain lat/lon/alt data
5. upload firmware-1.1.7.bin
6. perform initialization sequence (as above)
7. reset board and allow to sync GPS
8. from a second device, watch the packets again - "position" messages never contain lat/lon/alt data

Repeat steps 1-8 a couple times.

**More info**
- version 1.1.7 per commit 62a8c968e80c4e69755a7167232cddee023efe2e
- version 1.1.6 lost exact point, probably 4dd50df810780b6395701d3f10d31aea9247af8d

**Additionally**
Could have been just a coincidence, but the GPS lock acquisition appeared to be slower with firmware 1.1.7. Not enough data to tell for sure.

a-f-G-U-C, 2020-11-05:
Could not reproduce on further testing - possibly not a real issue. 

Closing with apologies.

closed 2020-11-05 by a-f-G-U-C

geeksville, 2020-11-11:
no problem.  thanks for the through effort and great writeup.