meshtastic/firmware #508: Question: Heltec Lora support

abhatikar, 2020-11-01:
Hello,

Would Meshtastic work on Heltic wireless stick https://heltec.org/project/wireless-stick/ ?
Would the heltec firmware prebuilt binary work on it ?

mc-hamster, 2020-11-01:
Probably not. The screen is a totally different resolution. 

The rest of the device should work with a few tweaks.

closed 2020-11-01 by mc-hamster

abhatikar, 2020-11-02:
Any plans to support that device?

mc-hamster, 2020-11-02:
It can probably be made to work without the screen with fairly low effort, but resizing all of the UI just for that one resolution will be a bit of effort.

abhatikar, 2020-11-02:
Would it be possible for a short call to explain how the code is organized and if it's possible, I could volunteer to add the support for the board.

thanks,

mc-hamster, 2020-11-02:
Join the developer slack and I'll walk you through what I can.

https://join.slack.com/t/meshtasticdev/shared_invite/zt-i7g00myc-ZNBbmQt6FFWPAPXFkvSUaQ