meshtastic/firmware #525: TinyGPS crash report on devices with external GPS

geeksville, 2020-11-18:
stack trace:

```
stack:
0x400e4692: TinyGPSPlus::endOfTermHandler() at /home/kevinh/development/meshtastic/meshtastic-esp32/.pio/libdeps/tlora-v2-1-1.6/TinyGPSPlus/src/TinyGPS++.cpp:152
0x400e4716: TinyGPSPlus::encode(char) at /home/kevinh/development/meshtastic/meshtastic-esp32/.pio/libdeps/tlora-v2-1-1.6/TinyGPSPlus/src/TinyGPS++.cpp:152
0x400d58a1: NMEAGPS::whileIdle() at /home/kevinh/development/meshtastic/meshtastic-esp32/src/gps/NMEAGPS.cpp:108
0x400d563f: GPS::runOnce() at /home/kevinh/development/meshtastic/meshtastic-esp32/src/gps/GPS.cpp:206
0x400d4722: concurrency::OSThread::run() at /home/kevinh/development/meshtastic/meshtastic-esp32/src/concurrency/OSThread.cpp:39
0x400e4929: ThreadController::runOrDelay() at /home/kevinh/development/meshtastic/meshtastic-esp32/.pio/libdeps/tlora-v2-1-1.6/Thread/ThreadController.cpp:153
0x400d8fe3: loop() at /home/kevinh/development/meshtastic/meshtastic-esp32/src/main.cpp:499
0x400f1e29: loopTask(void*) at /home/kevinh/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:19
~/development/meshtastic/meshtastic-esp32$ 
```

from https://meshtastic.discourse.group/t/gross-timeout-dead-start-0x44673-end-0x44673/1745

geeksville, 2020-11-18:
relevant portion of objdump:

```
400e4673:	78c2a2        	addi	a10, a2, 120
400e4676:	000246        	j	400e4683 <TinyGPSPlus::endOfTermHandler()+0x337>
400e4679:	d2a0b2        	movi	a11, 210
400e467c:	b2ba      	add.n	a11, a2, a11
400e467e:	c0a0a2        	movi	a10, 192
400e4681:	a2aa      	add.n	a10, a2, a10
400e4683:	ffc7a5        	call8	400e42fc <TinyGPSDecimal::set(char const*)>
400e4686:	3d2232        	l32i	a3, a2, 244
400e4689:	d2a052        	movi	a5, 210
400e468c:	e36316        	beqz	a3, 400e44c6 <TinyGPSPlus::endOfTermHandler()+0x17a>
400e468f:	3d2242        	l32i	a4, a2, 244
400e4692:	a3a8      	l32i.n	a10, a3, 40
400e4694:	a4b8      	l32i.n	a11, a4, 40
400e4696:	b11b81        	l32r	a8, 400d0b04 <_flash_cache_start+0xaec>
400e4699:	0008e0        	callx8	a8
400e469c:	e26a56        	bnez	a10, 400e44c6 <TinyGPSPlus::endOfTermHandler()+0x17a>
400e469f:	b388      	l32i.n	a8, a3, 44
400e46a1:	e20242        	l8ui	a4, a2, 226
400e46a4:	02a487        	bge	a4, a8, 400e46aa <TinyGPSPlus::endOfTermHandler()+0x35e>
400e46a7:	ff86c6        	j	400e44c6 <TinyGPSPlus::endOfTermHandler()+0x17a>
400e46aa:	069847        	bne	a8, a4, 400e46b4 <TinyGPSPlus::endOfTermHandler()+0x368>
400e46ad:	b25a      	add.n	a11, a2, a5
400e46af:	03ad      	mov.n	a10, a3
400e46b1:	ffc865        	call8	400e4338 <TinyGPSCustom::set(char const*)>
400e46b4:	c338      	l32i.n	a3, a3, 48
400e46b6:	fff486        	j	400e468c <TinyGPSPlus::endOfTermHandler()+0x340>
400e46b9:	120c      	movi.n	a2, 1
400e46bb:	f01d      	retw.n
400e46bd:	e1ba56        	bnez	a10, 400e44dc <TinyGPSPlus::endOfTermHandler()+0x190>
400e46c0:	ff8086        	j	400e44c6 <TinyGPSPlus::endOfTermHandler()+0x17a>
```

geeksville, 2020-11-18:
seems to be a bug in TinyGPS when they walk their customCandidates list.  We don't use custom candidates, so just turn all that code off instead of debugging it ;-)

dave032285, 2020-11-21:
> seems to be a bug in TinyGPS when they walk their customCandidates list. We don't use custom candidates, so just turn all that code off instead of debugging it ;-)

Could you point me in right direction for turning this off? You could probably tell i got no clue what im doing yet. Thank your for the constructive criticism its truly goes along way. The fact that you took the time to do so is motivational. Lol

geeksville, 2020-11-28:
@dave032285 I'll just merge this to master now.  If you could try it and report back that would help a lot.

commit df75cefe (2020-11-28): fix #525 - turn off (buggy) TinyGPS custom candidates support
closed 2020-11-28 by mc-hamster