meshtastic/firmware #526: "most recently read messageNum" is kept in MeshService

mc-hamster, 2020-11-18:
There is an ugly hack in there to shutdown the BLE API whenever someone connects through another instance of PhoneAPI.

This was a mistake I made originally - the notion of "most recently read messageNum" is kept in MeshService (which is a singleton used by everyone).  It really should be in PhoneAPI instead.

geeksville, 2021-01-17:
This issue has been mentioned on **Meshtastic**. There might be relevant details there:

https://meshtastic.discourse.group/t/webui-doesnt-show-devices-when-a-python-script-is-running/2186/2

garthvh, 2021-11-23:
Getting this bug fixed is a prerequisite to having multiple users with concurrent connections to a single device.

garthvh, 2021-11-23:
Based on some crash feedback from the iOS app it appears that some crashes that appear to be caused by doing a BLE PIN connection via the app may be caused by users trying to connect via serial and BLE being simultaneously.

ventz, 2021-11-24:
@garthvh Thanks for posting.

@geeksville I was that user :) -- I had a serial console opened on a T-Beam and was connecting via @garthvh's iOS app.

If it will be helpful to test anything specific, or provide any info/debug/etc, let me know.

geeksville, 2021-12-28:
This issue has been mentioned on **Meshtastic**. There might be relevant details there:

https://meshtastic.discourse.group/t/multiple-phones-connected-to-1-t-beam/4586/6

joshpirihi, 2022-01-30:
Does the device really need to keep a track of each client individually?  it'd make it really easy if every client got every FromRadio packet, whether they asked for it or not.  What issues might that cause?

mc-hamster, 2022-01-30:
Otherwise the client would get packets it has already seen and would need to manage state it wouldn't otherwise have to.

joshpirihi, 2022-01-30:
I missed the discussion on the call this morning --  I'll stay out of it until I know more about how hard this is

mc-hamster, 2022-01-30:
> I missed the discussion on the call this morning -- I'll stay out of it until I know more about how hard this is

Don't hold back and please keep questions coming. It's better that we learn more.

thebentern, 2022-04-26:
@garthvh do we know if this has been resolved with the transition to the new bluetooth stack?

garthvh, 2022-05-21:
This is not resolved, to be honest I think this in an enhancement not a bug really.  This is enabling multi user on a single device, which I think we kind of kicked way off down the road by increasing the node count.  @mc-hamster is probably more authoritative on this but I suspect multi user with 80 node limit is a hard problem for the nodedb as configured to solve.

garthvh, 2022-10-30:
Closed as not practical.

closed 2022-10-30 by garthvh

k1n6b0b, 2024-03-19:
There's a lot of discussion around use cases for multi connections 

Could this be reevaluated or is it never possible?

My post from another thread---

We would use this if it became available. We fit into the multiple groups / one device per group when doing remote activities like hiking, camping, especially remote music festivals. Right now we designate one person to be the conduit and they have to proxy all coordinations. This group is price sensitive, not kidding, we’d prefer not to have to spend 20-80/pp to add this connectivity.

Any increase in # concurrent connections via wifi or Bluetooth from the mobile app to the LoRa communicator would be huge for us. We primarily use the T-Echos but have one T-Beam as a basestation at camp

Communicating as the same named user is fine. Multiple users would be ideal. But multi simultaneous connections in any fashion is the request!

Wish I could contribute in some way to this ask :pray:

Thanks!

mammuthus, 2026-02-28:
It’s 2026 and Meshtastic still effectively supports only **one client per node**, and this is somehow classified as an “enhancement”, not a bug.
Issue [meshtastic/firmware#526](https://github.com/meshtastic/firmware/issues/526) has been around since 2020. The problem is well known: PhoneAPI / MeshService is a singleton, state is global, and multiple clients (TCP/BLE/Serial) just fight each other — one wins, the others silently break. For real‑world use (monitoring exporter + phone app + maybe automation) this is a hard limitation, **not a nice‑to‑have**: you have to choose, either your metrics collector gets the TCP slot, or your phone does.

At minimum this should be clearly documented (“one client only, others will break”). But after five years this deserves to be treated as a core bug/feature gap, not something that gets kicked down the road.