meshtastic/firmware #535: Heltec board stays asleep while connected to USB power

plughie, 2020-11-28:
Please - if you just have a question (i.e. not a bug report or a feature proposal), post in our [forum](https://meshtastic.discourse.group/) instead.

**Describe the bug**
The spec says that the radio is supposed to stay on when connected to USB.

**To Reproduce**
Steps to reproduce the behavior:
1. Plug device in to power. Connect to Meshtastic Android.

**Expected behavior**
The spec says that the radio is supposed to stay on when connected to USB. The Meshtastic app says it's sleeping.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Device info:**
Heltec board running 1.1.8

**Smartphone information (if relevant):**
 - Galaxy Tab 4

**Additional context**
Add any other context about the problem here.

mc-hamster, 2020-11-28:
I'll try this.

Hi Plugh! Long time!

geeksville, 2020-11-28:
btw: I bet the fix for this would be to change AnalogBatteryLevel.chargingVolt based on what actually shows up on a heltec device while USB power is connected.  I don't have one here in Taiwan to try though.

geeksville, 2020-12-10:
This issue has been mentioned on **Meshtastic**. There might be relevant details there:

https://meshtastic.discourse.group/t/want-to-help-alpha-test-this-build-1-1-20/1862/13

folletto95, 2020-12-10:
I'am not sure this is the same bug?

My board it's a TTGO Lora32 board, this is a new board with the new "firmware-tlora-v1-EU865-1.1.20.bin" .

In my case I flash the board with "ESPHome-Flasher-1.2.0-windows-x64.exe" and the new firmware.

I never connect an Android app at this board.

The thing's that dosen't  work it's the web interface like in the sceenshot:

[image]

Also the board didn't want to stay connect ad the near wifi acesspoint, after some time the board self disconnect from the wifi, only a reset or a hard reboot can riconnect.

geeksville, 2020-12-13:
This issue has been mentioned on **Meshtastic**. There might be relevant details there:

https://meshtastic.discourse.group/t/want-to-help-alpha-test-this-build-1-1-20/1862/39

mc-hamster, 2020-12-15:
@plughie thanks for putting a board in the mail. I’ll work on this bug soon as I get your board and will send it right back to you.

mc-hamster, 2020-12-15:
Schematic for the heltec:

https://github.com/Heltec-Aaron-Lee/WiFi_Kit_series/blob/master/SchematicDiagram/WiFi_LoRa_32(V2)/WIFI_LoRa_32_V2(868-915).PDF

I also see that the voltage divider is not what we expect it to be.

mc-hamster, 2020-12-15:
In addition, the lithium charger on this board ( TP4054 ) has a float voltage of 4.2v nominal, 4.242 max and 4.15 min. That means our assumption of fullVolt and chargingVolt isn't at the proper thresholds.

folletto95, 2020-12-15:
I modified my board, I cut tre 5v trace that connect te usb to the board and connected an external tp4056 whit a load pad, then I attached the load output on the 5v line of the lora32 board. I also add a voltage divider from 5v gpio35 and ground via a 2 100kohm resistors. I have recompiled the code whit the gpio35 set to battery voltage, now the board see something, but it shows always as usb powered.

folletto95, 2020-12-15:
From this morning the board seems that never lost the wifi connection

mc-hamster, 2020-12-21:
Got @plughie 's heltec board today. It is the same hardware as:

http://www.lilygo.cn/prod_view.aspx?TypeId=50003&Id=1134&FId=t3:50003:3

mc-hamster, 2020-12-22:
@geeksville I've tagged you as a reviewer for PR #579 . With the days of effort you put in that area, I'm not comfortable making a change without your eyes.

I'll wire in a battery later to check on AnalogBatteryLevel.chargingVolt. That might be a different problem.

closed 2020-12-23 by mc-hamster