meshtastic/firmware #536: Request: Hard code a location for non-gps units

plughie, 2020-11-28:
**Describe the feature request**
I have a couple of Heltec units without GPS that I would like to report from a static, hardcoded position. These will be fixed relay nodes with no use for a GPS, which would just draw extra unnecessary power.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Device info:**
 - Heltec units running 1.1.8 firmware

**Smartphone information (if relevant):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - App Version [e.g. 0.7.2]

**Additional context**
Add any other context about the problem here.

folletto95, 2020-11-30:
What app did you use for ios?

plughie, 2020-11-30:
I used Android, but app and OS versions are irrelevant because relay nodes will not be app connected.

IZ1IVA, 2020-12-07:
> **Describe the feature request**
> I have a couple of Heltec units without GPS that I would like to report from a static, hardcoded position. These will be fixed relay nodes with no use for a GPS, which would just draw extra unnecessary power.

I second that request, excellent idea!

geeksville, 2020-12-09:
thanks good idea.  working on this now.

geeksville, 2020-12-09:
In.  Run "meshtastic --setlat 32.7767 --setlon -96.7970 --setalt 1337".  If someone in this thread and has time and interest, it would be super great if you could send in a pull request to create the beginnings of "docs/python-cmd-guide.md" documenting these new flags (and eventually other things)

closed 2020-12-09 by geeksville

IZ1IVA, 2020-12-09:
> In. Run "meshtastic --setlat 32.7767 --setlon -96.7970 --setalt 1337". If someone in this thread and has time and interest, it would be super great if you could send in a pull request to create the beginnings of "docs/python-cmd-guide.md" documenting these new flags (and eventually other things)

Done. I presume --setalt expects an altitude in feet, doesn't it?

IZ1IVA, 2020-12-09:
Turns out it's metres. Or meters. Good!

IZ1IVA, 2020-12-10:
I'm having an issue here... if I manually set the position of a LoRa32 board, meshtastic --info doesn't return any position at all. Indeed, this node is listed in the mesh but doesn't show up in the map.