meshtastic/firmware #539: Device crash with wrong Wi-Fi Credentials

sachaw, 2020-11-28:
Seems to occur when wrong SSID is provided, have not tested with correct SSID and wrong PSK

**Video of behavior:** 
https://youtu.be/wMo1sfOUu1k

**Debug Log:**
```
... Reconnecting to WiFi access point
WiFi lost connection. Reason: 201
ignoring received packet due to error=-7
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:390] _eventCallback(): Reason: 201 - NO_AP_FOUND
************ [WiFi-event] event: 5 ************
Disconnected from WiFi access point
... Reconnecting to WiFi access pointGuru Meditation Error: Core  1 panic'ed (Cache disabled but cached memory region accessed)
Core 1 register dump:
PC      : 0x400812b4  PS      : 0x00060034  A0      : 0x800812e0  A1      : 0x3ffbede0  
A2      : 0x00000001  A3      : 0x3ffc4a14  A4      : 0x800926e3  A5      : 0x3ffbede0  
A6      : 0x00000020  A7      : 0x00000020  A8      : 0xbad00bad  A9      : 0x3f419abd  
A10     : 0x3ffd3834  A11     : 0x00000001  A12     : 0x00000000  A13     : 0x3ffbedc0  
A14     : 0x00000003  A15     : 0x00060023  SAR     : 0x0000001a  EXCCAUSE: 0x00000007  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  
Core 1 was running in ISR context:
EPC1    : 0x4008695f  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x400812b4
ELF file SHA256: 0000000000000000
Backtrace: 0x400812b4:0x3ffbede0 0x400812dd:0x3ffbee10 0x40081359:0x3ffbee30 0x4008246d:0x3ffbee50 0x4008695c:0x3ffba1b0 0x40084565:0x3ffba1d0
Rebooting...

```

mc-hamster, 2020-11-28:
I was able to reproduce after 18 hours of up time. Not that bad but not great ether. Will keep an eye on this.

mc-hamster, 2021-03-03:
have not seen this issue reoccur or am able to reproduce it.

closing for now.

if it happens again, we can reopen this.

closed 2021-03-03 by mc-hamster