meshtastic/firmware #540: Occasionally reset the time based on GPS

geeksville, 2020-12-03:
Many boards (in particular TBEAMs) don't have a 32khz xtal.  Therefore they are just using an RC clock for timekeeping and that drifts pretty fast.

So the time of day clock (not millis() though) should be reset occasionally if !defined(HAS_HIGH_ACCURACY_CLOCK).

geeksville, 2020-12-03:
This issue has been mentioned on **Meshtastic**. There might be relevant details there:

https://meshtastic.discourse.group/t/settime-command-periodically-via-bluetooth/1807/6

mc-hamster, 2020-12-03:
The esp32 has internal methods to improve clock accuracy:

https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/system_time.html

geeksville, 2020-12-25:
Most of this is fixed by #584 (though we should still reset the clock based on GPS occasionally (once a day?)

commit 21570fc2 (2020-12-27): fix #540 use gps to fix rtc clock drift every 12 hrs
closed 2020-12-27 by geeksville