meshtastic/firmware #541: Various http server issues on first setup of WiFi AP, after device flash

ry-w, 2020-12-03:
**Describe the bug**
After first setting the device to Wifi AP using the meshtastic command line tool, the device does not start the http server properly.  There are a few other related issues here as well, including a log typo and device screen status.

**To Reproduce**
1. Flash firmware 
2. Setup wi-fi access point then view logs on device
>    meshtastic --set wifi_ap_mode true --setstr wifi_ssid mywifissid --setstr wifi_password mywifipsw
>     meshtastic

Output:
> Connected to radio
>  Setting wifi_ap_mode to True
>  Setting wifi_ssid to mywifissid
>  Setting wifi_password to mywifipsw
>  Writing modified preferences to device

2. a. [**First issue**] Logs don't indicate that device is started

3. Press reboot button
4. On bootup, once wifi SSID appears to the client computer, it can be connected to
4. a. [**Second issue**] HTTP services are not available to the client computer, even though it can connect.  Logs show that [WiFi-event] event: 14 attempts to start 3 times after what appears to be an initial successful start.  'setupSockets' fails in the last 2 attempts.
4. b. [**Third issue**] Device screen shows 'Unknown Status', on far-right 'dot' screen.  It shows: 
>  WiFi: Software AP
>  Unknown Status
>  SSID / PWD [flash between the two]
>  http://meshtastic.local

**Another minor issue**:  Log typo, 'sistening':
>  API server sistening on TCP port 4403

5. After some time, press reboot button again.
6. Logs now show that device starts event 14 once properly, and the server will now respond to the client computer.

**Expected behavior**
1. (2. a.), either a device reboot or startup of wifi should show in the logs.
2. (4. a.), wifi service expected to startup only once, particularly if the first attempt was successful.  Is it looping when setting up DNS?
3. (4. b.), wifi screen expected to show IP, ssid, and other relevant information including errors on startup.

**Device info:**
 - Device model: TBEAM v1.1:
 - Software Version: 1.1.8 (firmware-tbeam-US-1.1.8.bin)

**Device output from 'meshtastic' command line tool**
Installing AES128 key!
Wanted region 1, using US
legacy_region=1.0-US, region=1, NODENUM=0x1c66b338, dbsize=2
Turning on screen
Starting meshradio init...
Set radio: name=Default, config=3, ch=6, power=0
Radio myRegion->freq: 903.080017
Radio myRegion->spacing: 2.160000
Radio myRegion->numChannels: 13
Radio channel_num: 6
Radio frequency: 916.040039
Set radio: final power level=17
RF95 init result 0
Waiting for SSL Cert to be generated.
SSL Cert Ready!
[E][Preferences.cpp:457] getBytesLength(): nvs_get_blob len fail: PK NOT_FOUND
[E][Preferences.cpp:457] getBytesLength(): nvs_get_blob len fail: cert NOT_FOUND
Checking if we have a previously saved SSL Certificate.
Creating the certificate. This may take a while. Please wait...
Creating the certificate was successful
Created Private Key: 1191 Bytes
Created Certificate: 780 Bytes
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 0 - WIFI_READY
************ [WiFi-event] event: 0 ************
WiFi interface ready
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 14 - AP_START
************ [WiFi-event] event: 14 ************
WiFi access point started
192.168.4.1
Initializing Web Server ...
Checking if we have a previously saved SSL Certificate.
Retrieved Private Key: 1191 Bytes
Retrieved Certificate: 780 Bytes
Starting Web Servers...
HTTP and HTTPS Web Servers Ready! :-) 
API server sistening on TCP port 4403
STARTING WIFI AP: ssid=mywifissid, ok=1
MY IP ADDRESS: 192.168.42.1
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 14 - AP_START
************ [WiFi-event] event: 14 ************
WiFi access point started
192.168.42.1
Initializing Web Server ...
Checking if we have a previously saved SSL Certificate.
Retrieved Private Key: 1191 Bytes
Retrieved Certificate: 780 Bytes
Starting Web Servers...
setupSockets failed
HTTP and HTTPS Web Servers Failed! ;-( 
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 14 - AP_START
************ [WiFi-event] event: 14 ************
WiFi access point started
192.168.42.1
Initializing Web Server ...
Checking if we have a previously saved SSL Certificate.
Retrieved Private Key: 1191 Bytes
Retrieved Certificate: 780 Bytes
Starting Web Servers...
mDNS responder setupSockets failed
started
mHTTP and HTTPS Web Servers Failed! ;-( 
DNS Host: Meshtastic.local

mc-hamster, 2020-12-13:
From Charles

`booted, wake cause 0 (boot count 1), reset_reason=reset
-------------------- Setting forceSoftAP = 1
Filesystem files:
  /static/app.js.gz
  /static/app.css.gz   
  /db.proto
  /static/index.html.gz
I2C device found at address 0x34
axp192 PMU found
I2C device found at address 0x3c
ssd1306 display found
done
Meshtastic swver=1.1.20, hwver=unset
Setting random seed 184356010
Total heap: 255924
Free heap: 223296
Total PSRAM: 4194252
Free PSRAM: 4194252
NVS: UsedEntries 257, FreeEntries 373, AllEntries 630
chip id detect 0x3
Detect CHIP :AXP192
OUTPUT Register 0x5f
AXP192 Begin PASS
DCDC1: ENABLE
DCDC2: ENABLE
LDO2: ENABLE
LDO3: ENABLE
DCDC3: ENABLE
Exten: ENABLE
----------------------------------------
DCDC1: ENABLE
DCDC2: ENABLE
LDO2: ENABLE
LDO3: ENABLE
DCDC3: ENABLE
Exten: ENABLE
SRC REG:0xcf
Charging enable is enable
Charging target-voltage : 0x2
 end when the charge current is lower than 10% of the set value
Charge current : 1000.00 mA
Battery: usbPower=1, isCharging=0, batMv=4140, batPct=100
Read RTC time as 0 (cur millis 217) quality=0
WANT GPS=1
Setting GPS power=1
Connected to UBLOX GPS successfully
Setting default preferences!
Installing AES128 key!
Wanted region 0, using Unset
Initial packet id 1626927516, numPacketId 4294967295
Loading saved preferences
Loaded saved preferences version 11
This build does not specify a HW_VERSION
Installing AES128 key!
Wanted region 1, using US
legacy_region=1.0-US, region=1, NODENUM=0x1c66b7b0, dbsize=3
Turning on screen
Starting meshradio init...
(bw=125, sf=12, cr=4/8) packet symLen=32 ms, payloadSize=16, time 2269 ms
Set radio: name=Default, config=3, ch=6, power=0
Radio myRegion->freq: 903.080017
Radio myRegion->spacing: 2.160000
Radio myRegion->numChannels: 13
Radio channel_num: 6
Radio frequency: 916.040039
Short packet time: 2269 msec
Set radio: final power level=17
RF95 init result 0
[... 105 lines cut ...]
Pre BT: 137864 heap size
Starting bluetooth
Screen: Powered...
Completed sending (id=0x60f8f19f Fr0xb0 To0xff, WantAck0, HopLim3 encrypted)
BLE task running
registered service 0x1800 with handle=1
registered service 0x1801 with handle=6
registered service 6ba1b218-15a8-461f-9fa8-5dcae273eafd with handle=10
registered service cb0b9a0b-a84c-4c0d-bdbb-442e3144ee30 with handle=18
BLE advertisting type=0, Private=0, Device Address: 55:10:52:1c:66:b7
Can not send yet, busyRx
Can not send yet, busyRx
Lora RX (id=0x60f8f19f Fr0xb0 To0xff, WantAck0, HopLim2 encrypted rxSNR=9)
Rx someone rebroadcasting for us (id=0x60f8f19f Fr0xb0 To0xff, WantAck0, HopLim2 encrypted rxSNR=9)
Found existing packet record for fr=0x1c66b7b0,to=0xffffffff,id=1626927519
Ignoring incoming msg, because we've already seen it: fr=0x1c66b7b0,to=0xffffffff,id=1626927519,hop_limit=2
Starting low level send (id=0x60f8f1a0 Fr0xb0 To0xff, WantAck0, HopLim3 encrypted)
Completed sending (id=0x60f8f1a0 Fr0xb0 To0xff, WantAck0, HopLim3 encrypted)
Starting low level send (id=0x60f8f1a1 Fr0xb0 To0x58, WantAck0, HopLim3 encrypted rxtime=1607831929)
Completed sending (id=0x60f8f1a1 Fr0xb0 To0x58, WantAck0, HopLim3 encrypted rxtime=1607831929)
Battery: usbPower=1, isCharging=0, batMv=4139, batPct=100
Lora RX (id=0x60f8f1a0 Fr0xb0 To0xff, WantAck0, HopLim2 encrypted rxSNR=10)
Rx someone rebroadcasting for us (id=0x60f8f1a0 Fr0xb0 To0xff, WantAck0, HopLim2 encrypted rxSNR=10)
Found existing packet record for fr=0x1c66b7b0,to=0xffffffff,id=1626927520
Ignoring incoming msg, because we've already seen it: fr=0x1c66b7b0,to=0xffffffff,id=1626927520,hop_limit=2`

mc-hamster, 2020-12-13:
From Charles

booted, wake cause 0 (boot count 1), reset_reason=reset
-------------------- Setting forceSoftAP = 1
Filesystem files:
  /static/app.js.gz
  /static/app.css.gz   
  /db.proto
  /static/index.html.gz
I2C device found at address 0x34
axp192 PMU found
I2C device found at address 0x3c
ssd1306 display found
done
Meshtastic swver=1.1.20, hwver=unset
Setting random seed 184356010
Total heap: 255924
Free heap: 223296
Total PSRAM: 4194252
Free PSRAM: 4194252
NVS: UsedEntries 257, FreeEntries 373, AllEntries 630
chip id detect 0x3
Detect CHIP :AXP192
OUTPUT Register 0x5f
AXP192 Begin PASS
DCDC1: ENABLE
DCDC2: ENABLE
LDO2: ENABLE
LDO3: ENABLE
DCDC3: ENABLE
Exten: ENABLE
----------------------------------------
DCDC1: ENABLE
DCDC2: ENABLE
LDO2: ENABLE
LDO3: ENABLE
DCDC3: ENABLE
Exten: ENABLE
SRC REG:0xcf
Charging enable is enable
Charging target-voltage : 0x2
 end when the charge current is lower than 10% of the set value
Charge current : 1000.00 mA
Battery: usbPower=1, isCharging=0, batMv=4140, batPct=100
Read RTC time as 0 (cur millis 217) quality=0
WANT GPS=1
Setting GPS power=1
Connected to UBLOX GPS successfully
Setting default preferences!
Installing AES128 key!
Wanted region 0, using Unset
Initial packet id 1626927516, numPacketId 4294967295
Loading saved preferences
Loaded saved preferences version 11
This build does not specify a HW_VERSION
Installing AES128 key!
Wanted region 1, using US
legacy_region=1.0-US, region=1, NODENUM=0x1c66b7b0, dbsize=3
Turning on screen
Starting meshradio init...
(bw=125, sf=12, cr=4/8) packet symLen=32 ms, payloadSize=16, time 2269 ms
Set radio: name=Default, config=3, ch=6, power=0
Radio myRegion->freq: 903.080017
Radio myRegion->spacing: 2.160000
Radio myRegion->numChannels: 13
Radio channel_num: 6
Radio frequency: 916.040039
Short packet time: 2269 msec
Set radio: final power level=17
RF95 init result 0
[... 105 lines cut ...]
Pre BT: 137864 heap size
Starting bluetooth
Screen: Powered...
Completed sending (id=0x60f8f19f Fr0xb0 To0xff, WantAck0, HopLim3 encrypted)
BLE task running
registered service 0x1800 with handle=1
registered service 0x1801 with handle=6
registered service 6ba1b218-15a8-461f-9fa8-5dcae273eafd with handle=10
registered service cb0b9a0b-a84c-4c0d-bdbb-442e3144ee30 with handle=18
BLE advertisting type=0, Private=0, Device Address: 55:10:52:1c:66:b7
Can not send yet, busyRx
Can not send yet, busyRx
Lora RX (id=0x60f8f19f Fr0xb0 To0xff, WantAck0, HopLim2 encrypted rxSNR=9)
Rx someone rebroadcasting for us (id=0x60f8f19f Fr0xb0 To0xff, WantAck0, HopLim2 encrypted rxSNR=9)
Found existing packet record for fr=0x1c66b7b0,to=0xffffffff,id=1626927519
Ignoring incoming msg, because we've already seen it: fr=0x1c66b7b0,to=0xffffffff,id=1626927519,hop_limit=2
Starting low level send (id=0x60f8f1a0 Fr0xb0 To0xff, WantAck0, HopLim3 encrypted)
Completed sending (id=0x60f8f1a0 Fr0xb0 To0xff, WantAck0, HopLim3 encrypted)
Starting low level send (id=0x60f8f1a1 Fr0xb0 To0x58, WantAck0, HopLim3 encrypted rxtime=1607831929)
Completed sending (id=0x60f8f1a1 Fr0xb0 To0x58, WantAck0, HopLim3 encrypted rxtime=1607831929)
Battery: usbPower=1, isCharging=0, batMv=4139, batPct=100
Lora RX (id=0x60f8f1a0 Fr0xb0 To0xff, WantAck0, HopLim2 encrypted rxSNR=10)
Rx someone rebroadcasting for us (id=0x60f8f1a0 Fr0xb0 To0xff, WantAck0, HopLim2 encrypted rxSNR=10)
Found existing packet record for fr=0x1c66b7b0,to=0xffffffff,id=1626927520
Ignoring incoming msg, because we've already seen it: fr=0x1c66b7b0,to=0xffffffff,id=1626927520,hop_limit=2

crossan007, 2020-12-13:
I tried using PlatformIO to `Erase Flash` and re-upload the build from your dev branch 9f9573d2, and even with a clean install I get the same failure mode.

crossan007, 2020-12-13:
Might be related to https://github.com/espressif/arduino-esp32/issues/3006

crossan007, 2020-12-13:
Also maybe related: https://github.com/espressif/arduino-esp32/issues/3444

I wonder if meshtastic needs to decouple "starting" the HTTP(s) servers from the "WiFi ready" event;

crossan007, 2020-12-13:
I was somehow able to get past this; possibly by disabling the WiFi on my phone so that it didn't try connecting the instant the ESP AP became available, but I can't confirm whether that was actually the case.

mc-hamster, 2020-12-13:
With @ry-w and @crossan007's information, I'm able to reproduce this.

mc-hamster, 2020-12-13:
I don't understand the cause but I see the symptom.

closed 2020-12-13 by mc-hamster

mc-hamster, 2020-12-13:
> I was somehow able to get past this; possibly by disabling the WiFi on my phone so that it didn't try connecting the instant the ESP AP became available, but I can't confirm whether that was actually the case.
> 
> hypothesis is that having phones WiFi "on" caused it to immediately connect to the ESP; and multi-trigger that wifi event 14 (thus breaking HTTP(s) services)

I suspect a periodic race condition in the Espressif SDK.

commit 86952c54 (2020-12-13): Fixed #541 - Work around to bug in espressif softap event handler