meshtastic/firmware #546: Feature Request: Channel specified device defaults

mc-hamster, 2020-12-10:
Right now, we have device settings that are set by us. Let's call this "factory defaults". We also have device settings that can be set by the end user on a per device basis. Let's call this "device settings".

I'd like to propose "channel device settings". 

The user story would be:

So, as the originator of a channel, if other devices join my mesh, I want them to inherit my settings so that I don't need to individually configure all of the end user devices.

The devices could, of course, choose to use different settings.

For my use case, I now have 14 radios and really am not looking forward to the idea of individually configuring each device.

We may need to have some level of access control of the channel device settings, but that would go into another ticket.

clavo2, 2020-12-10:
This is really needed.  As of now there is no easy way to make the map updating faster so that it's useful without the API.

crossan007, 2020-12-12:
Some additional thoughts on "features" to include here:

*  WiFi SSID/PSK synchronization (facilitate all LoRa nodes to connect to a distributed list of SSIDs)
*  WiFi AP synchronization (configure all LoRa nodes to broadcast the same SSID for WiFi clients to connect to)

crossan007, 2020-12-12:
Also a _possible_ candidate would be an "authoritative" contents for SPIFFS that could be sync'd over mesh

Given the mesh throughput constraints, this might not really be feasible; just posing the idea.

mc-hamster, 2021-03-27:
Closing this request. It can be done now with the admin channel.

closed 2021-03-27 by mc-hamster