meshtastic/firmware #549: SPIFFS Auto Management

crossan007, 2020-12-12:
Given a device where the 1.1.20+ firmware has been flashed without the corresponding SPIFFS partition, the device (if connected to an internet-connected WiFi network) should automatically download the latest SPIFFS partition and prepare it for use.

This would give us a way to keep the web interface up-to-date, as well as to head-off support requests by folks that didn't use the `device-install.sh` scripts

crossan007, 2020-12-12:
I would suggest this look at the GitHub URL for the latest release, and pull whatever spiffs bin is appropriate for the platform (not knowing how any of the mechanics actually work for interacting with spiffs)

mc-hamster, 2021-03-27:
Similar feature can be done now with the PyGUI that Zebus is working on.

Closing this request.

closed 2021-03-27 by mc-hamster