meshtastic/firmware #554: PowerFSM management in meshhttp

crossan007, 2020-12-12:
When a WebUI is connected to the radio, it may or may not be desirable for the radio to enter low-power mode.

This choice should be up to the user.

I don't see any references to `powerFSM` in `meshhttp.cpp`, and have noticed that the radio goes to low power mode when a WebUI is connected (unless the radio is plugged into USB power).

I would like to present through the WebUI an option to prevent the radio from sleeping (even if on battery), but don't see a mechanism to do so.

Proposed solution:
Implement an activity timer within (meshhttp?) which counts seconds since last HTTP request;  if the count exceeds a user-configurable threshold (maybe exposed as a UserSetting setting via protobuf?), then the device is allowed to sleep;  otherwise, it should stay "on" and continue to service HTTP requests

mc-hamster, 2020-12-12:
Do you have access to the wait_bluetooth_secs preference? That's the setting for the timer that counts down and turns off the radio. If you turn up that value (the default is 60 seconds) then you can set the radio not to turn off.

crossan007, 2020-12-12:
are HTTP interactions counted towards this timer?

mc-hamster, 2020-12-22:
they are not, but that's a good point. they should be.

Noki, 2020-12-22:
Probably an always on mode (disabling all the low power optimizations) for devices directly connected to the powergrid, would be a good idea, especially with plans to extend the mesh to other nodes via the internet.

mc-hamster, 2020-12-23:
> Probably an always on mode (disabling all the low power optimizations) for devices directly connected to the powergrid, would be a good idea, especially with plans to extend the mesh to other nodes via the internet.

Existing logic is to have the device be "always on" if powered from USB.

mc-hamster, 2020-12-23:
> are HTTP interactions counted towards this timer?

@crossan007 While you're developing the web interface, please use the change I just checked into my fork. It has the changes for this feature request. You're in a better position to give feedback on the device behavior with this change.

Noki, 2020-12-23:
@mc-hamster I came to my conclusion because I observed the behavior that the webserver went away without a http connection (#554 not applied) and does not come up again. The serial log shows the following:

[image]

This might be totally unrelated. But I would expect the webserver to be always running on usbPower and bringing it back up from sleep should not fail.

Bringing it back up after a hard-reset works just fine:

[image]

mc-hamster, 2020-12-23:
@Noki i bet you ran into this bug:

https://github.com/meshtastic/Meshtastic-device/issues/407

closed 2020-12-25 by mc-hamster