meshtastic/firmware #556: fixed_position only takes effect when there is no GPS Lock

crossan007, 2020-12-12:
It may be desired taht an administratively-set `fixed_position` be preferred over the position reported by any on-board GPS hardware.

It looks like `MeshService.cpp` will always prefer the GPS hardware-reported position _if it has a lock_ regardless of whether `fixed_position` is configured as:
```
Position pos = Position_init_default;

    if (gps->hasLock()) {
        if (gps->altitude != 0)
            pos.altitude = gps->altitude;
        pos.latitude_i = gps->latitude;
        pos.longitude_i = gps->longitude;
    }
    else {
        // The GPS has lost lock, if we are fixed position we should just keep using
        // the old position
        if(radioConfig.preferences.fixed_position) {
            NodeInfo *node = nodeDB.getNode(nodeDB.getNodeNum());
            assert(node);
            assert(node->has_position);
            pos = node->position;
            DEBUG_MSG("WARNING: Using fixed position\n");
        }
    }

```

mc-hamster, 2020-12-23:
@geeksville What do you think of this change in behavior? I can swap the order if you're ok with the proposal.

mc-hamster, 2021-02-07:
fixed_position is a boolean. what it means has been received by the gps (or phone?), don't discard it when the gps signal is lost.

closing bug.

closed 2021-02-07 by mc-hamster