meshtastic/firmware #561: WiFi SSID Scan

crossan007, 2020-12-13:
Create an HTTP API which returns a JSON representation of the WiFi SSIDs visible from the current device

mc-hamster, 2020-12-13:
This is done.

Your endpoint is:
  /json/scanNetworks

It's perfectly valid to call /json/scanNetworks more than once. Networks don't always broadcast and the scanner may not catch it the first time.

It's also valid to have the same network name show up multiple times. That's a case where there's the same SSID but different BSSID. I don't have access to the BSSID.

mc-hamster, 2020-12-13:
Known problems:

I'm not escaping double quotes, I should do that.
Unicode characters may have problems. I don't know how that would work.

closed 2020-12-13 by mc-hamster