meshtastic/firmware #567: Support NTP for WiFi connected non-GPS nodes

crossan007, 2020-12-14:
For cases where a TBeam may be "indoors" (and not have strong GPS reception) and connected to wifi and be geographically "static" (not moving, and likely to have a hardcoded latitude/longitude set), the device should support syncing time from an upstream NTP server so that it can have a "correct time"

sachaw, 2020-12-15:
+1 for NTP support, the server can be obtained through DHCP option 004 or 042.

Black6spdZ, 2021-01-08:
just code in a few pool.ntp servers

mc-hamster, 2021-12-28:
The GPS time is used for two cases ... it's used by the local device to be able to timestamp its own events and it's also used by the local device to transmit time out to the mesh network to be used by other nodes that may not have a GPS to set its own time. The GPS time is assumed to be as accurate as we can get.

If NTP is implemented, I'd suggest not sending this out to the mesh network and that time only be kept local.

mc-hamster, 2021-12-28:
Work here would be to modify RTCQuality in RTC.h to:

enum RTCQuality {
    /// We haven't had our RTC set yet
    RTCQualityNone = 0,

    /// Some other node gave us a time we can use
    RTCQualityFromNet = 1,

    /// Our time is based on NTP
    RTCQualityNTP= 2

    /// Our time is based on our own GPS
    RTCQualityGPS = 3
};

and then call perhapsSetRTC(RTCQualityNTP, t) when we get time from an NTP server.

If we get time from an NTP server, suggest refreshing that time every 24 hours to reduce clock drift.

mc-hamster, 2021-12-29:
I put in some effort to implement... the arduino/ntpclient library is way too buggy (not maintained either). espressif has a ntp client built into the sdk but that client updates the device rtc directly. We need to be the one to update the rtc and not have it abstracted away for us.

Need to find another ntp client.

mc-hamster, 2022-01-04:
The bug in the NTP client is it will set the time even if it can't connect to the NTP server. Work around was to check to see if a connection was made.

mc-hamster, 2022-01-04:
this is done.

closed 2022-01-04 by mc-hamster