meshtastic/firmware #576: JSON SPIFFS response has extra object

crossan007, 2020-12-21:
The object returned from /json/spiffs/browse/static has an extra blank object in the second array position (1-indexed)

[image]

crossan007, 2020-12-21:
```
{
"data": {
"files": [{
"nameModified": "static/app.js","name": "static/app.js.gz","size": 77980},
{
},
{
"nameModified": "static/index.html","name": "static/index.html.gz","size": 298},
{
"nameModified": "static/app.css","name": "static/app.css.gz","size": 1059},
{
"name": "static/app.js","size": 19200}],"filesystem" : {"total" : 414401,"used" : 101404,"free" : 312997}
},
"status": "ok"
}

```

mc-hamster, 2020-12-21:
Thanks for the bug report!

closed 2020-12-21 by mc-hamster