meshtastic/firmware #586: Sometimes TBEAMs can sleep even though they are on USB power

geeksville, 2020-12-24:
reported by @IZ1IVA in linked thread

geeksville, 2020-12-24:
This issue has been mentioned on **Meshtastic**. There might be relevant details there:

https://meshtastic.discourse.group/t/continuous-message-sending-through-terminal/1890/33

4DA, 2021-04-06:
I have the same issue with ttgo LORA32 (V2.1 1.6). Device starts in "ON" power mode and goes down power saving chain quickly.

meshtastic --info
```
My info: { "myNodeNum": 1618960128, "numBands": 8, "firmwareVersion": "1.2.23", "rebootCount": 22, "messageTimeoutMsec": 300000, "minAppVersion": 20200, "maxChannels": 8 }
```
log:
```
??:??:?? 2 PowerFSM init, USB power=0
[D][esp32-hal-cpu.c:189] setCpuFrequencyMhz(): PLL: 320 / 4 = 80 Mhz, APB: 80000000 Hz
Connected to radio
Transition powerFSM transition=boot timeout, from=BOOT to=ON
Pre BT: 209820 heap size
[D][esp32-hal-i2c.c:1345] i2cProcQueue():  Gross Timeout Dead start=0x17f1, end=0x17f1, =0, max=50 error=1
[E][esp32-hal-i2c.c:318] i2cDumpI2c(): i2c=0x3ffbe2e8
[I][esp32-hal-i2c.c:319] i2cDumpI2c(): dev=0x60013000 date=0x16042000
[I][esp32-hal-i2c.c:321] i2cDumpI2c(): lock=0x3ffbc28c
[I][esp32-hal-i2c.c:323] i2cDumpI2c(): num=0
[I][esp32-hal-i2c.c:324] i2cDumpI2c(): mode=1
[I][esp32-hal-i2c.c:325] i2cDumpI2c(): stage=3
[I][esp32-hal-i2c.c:326] i2cDumpI2c(): error=1
[I][esp32-hal-i2c.c:327] i2cDumpI2c(): event=0x3ffbadac bits=10
[I][esp32-hal-i2c.c:328] i2cDumpI2c(): intr_handle=0x3ffbaddc
[I][esp32-hal-i2c.c:329] i2cDumpI2c(): dq=0x3ffd9460
[I][esp32-hal-i2c.c:330] i2cDumpI2c(): queueCount=1
[I][esp32-hal-i2c.c:331] i2cDumpI2c(): queuePos=0
[I][esp32-hal-i2c.c:332] i2cDumpI2c(): errorByteCnt=2
[I][esp32-hal-i2c.c:333] i2cDumpI2c(): errorQueue=0
[I][esp32-hal-i2c.c:334] i2cDumpI2c(): debugFlags=0x00000000
[I][esp32-hal-i2c.c:311] i2cDumpDqData(): Debug Buffer not Enabled
[I][esp32-hal-i2c.c:354] i2cDumpInts(): Debug Buffer not Enabled
Transition powerFSM transition=Screen-on timeout, from=ON to=DARK
Transition powerFSM transition=Bluetooth timeout, from=DARK to=LS
Shutdown BT: 224192 heap siz
```

4DA, 2021-04-09:
I've debugged a stuff a little bit, here are the battery voltage readings (battery is not connected):

getBattVoltage() = 3641.601562
getBattVoltage() = 3641.601562
getBattVoltage() = 3635.156250
getBattVoltage() = 3615.820312

Does that mean I have a faulty board?

mc-hamster, 2022-01-17:
closed. no longer an issue.

closed 2022-01-17 by mc-hamster