meshtastic/firmware #587: Build error

cronyx, 2020-12-24:
Hi, I was trying to build firmware from source according to the build instructions and received an error.
[image]

geeksville, 2020-12-25:
oh interesting!  It seems like the python in your platformio is old.  Can you check if you are running the latest version of platformio?  I think that "pio upgrade" will do this?

geeksville, 2020-12-25:
this might help: https://docs.platformio.org/en/latest/faq.html#install-python-interpreter

geeksville, 2020-12-25:
but just in case, I pushed a version of  our build script that I think will also work on python 2.x.  can you give it a try? (it will go into master in about an hr)

cronyx, 2020-12-25:
> this might help: https://docs.platformio.org/en/latest/faq.html#install-python-interpreter

Hi, platformio version is 5.0.3
```
You're up-to-date!
PlatformIO 5.0.3 is currently the newest version available.
```

Python versions:
```
python -V
Python 2.7.12
python3 -V
Python 3.5.2
```

> but just in case, I pushed a version of our build script that I think will also work on python 2.x. can you give it a try? (it will go into master in about an hr)

Thanks, it work, but need some corrections:
```
# General options that are passed to the C and C++ compilers
projenv.Append(CCFLAGS=[
    "-DAPP_VERSION=" + verStr
    ])
```

commit 2edc6b36 (2020-12-27): fix #587 thanks @cronyx
closed 2020-12-27 by geeksville