meshtastic/firmware #588: Calculate air time duty cycles

mc-hamster, 2020-12-25:
Calculate air time duty cycles

geeksville, 2020-12-25:
This issue has been mentioned on **Meshtastic**. There might be relevant details there:

https://meshtastic.discourse.group/t/ip-over-meshtastic-kinda-implemented/1775/12

geeksville, 2020-12-25:
I really like the "track over a long period of time idea (a day or two)"  so that throttling (when eventually added) doesn't kick in with false positives.

geeksville, 2020-12-25:
also just knowing what 'real world duty cycle' we are currently getting would be useful.

mc-hamster, 2020-12-25:
I think a first pass would be to report the TX airtime. A next step may be to also report on RX airtime so we have an idea of cannel utilization.

geeksville, 2020-12-25:
ooh really good point wrt rx airtime telling us % channel utilization also.

geeksville, 2020-12-25:
Also nodes with "higher rx airtime than others" might be a *really* good heuristic for "nodes that should probably be preferentially considered for  routing (someday when we are doing something smarter than naive flood routing)"

mc-hamster, 2020-12-25:
Note to self: https://github.com/meshtastic/Meshtastic-device/commit/1839f8f7ca476fa5b094bbd5544811f6532867bf

mc-hamster, 2020-12-25:
@geeksville Thinking about channel utilization, I think there are two measures.

1) The utilization the channel are we using for our current network
2) The utilization of the lora channel by the current device, the devices on our current meshtastic channel and other lora devices that may have similar channel configurations.

I'll assume that for #2, if our radio receives a packet that is not designated for the device, it's discarded. Where does that happen? I'd like to also include that.

crossan007, 2020-12-25:
Seems like we need 4 counters?

Transmitted packets (originating on device)
Transmitted packets (forwarded for other devices)
Received packets (destined for device)
Received packets (for other devices)

mc-hamster, 2020-12-26:
> Seems like we need 4 counters?
> 
> Transmitted packets (originating on device)
> Transmitted packets (forwarded for other devices)
> Received packets (destined for device)
> Received packets (for other devices)

Hmm ...

Other than for reporting, how would having a counter for packets originating from the device vs to be forwarded to other devices be used?

When maintaining a 48 hour history, each counter costs us about ~100 bytes of RAM.

Regardless, adding more counters will be easy if it's needed. :)

closed 2020-12-31 by mc-hamster