meshtastic/firmware #597: watchdog timer exception while building https certificate

geeksville, 2020-12-27:
Not urgent, because it did the right thing on the second attempt.  But I think we need to bump up our minimum task watchdog timeout.  Or start the task watchdog only after HTTPS certs were generated.

```
~/development/meshtastic/meshtastic-esp32$ bin/exception_decoder.py -e .pio/build/tbeam/firmware.elf ex
stack:
0x4008ff5f: invoke_abort at /home/kevinh/development/meshtastic/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:715
0x400902e5: abort at /home/kevinh/development/meshtastic/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:715
0x40112e50: esp_task_wdt_init at /home/kevinh/development/meshtastic/esp32-arduino-lib-builder/esp-idf/components/esp32/task_wdt.c:252
0x4008246d: _xt_lowint1 at /home/kevinh/development/meshtastic/esp32-arduino-lib-builder/esp-idf/components/freertos/xtensa_vectors.S:1154
0x401d8cfc: mpi_mul_hlp at /home/kevinh/development/meshtastic/esp32-arduino-lib-builder/esp-idf/components/mbedtls/mbedtls/library/bignum.c:1463 (discriminator 2)
0x4012af92: mpi_montmul at /home/kevinh/development/meshtastic/esp32-arduino-lib-builder/esp-idf/components/mbedtls/mbedtls/library/bignum.c:2572
0x4012bd22: mbedtls_mpi_exp_mod at /home/kevinh/development/meshtastic/esp32-arduino-lib-builder/esp-idf/components/mbedtls/mbedtls/library/bignum.c:2572
0x40134944: mbedtls_rsa_private at /home/kevinh/development/meshtastic/esp32-arduino-lib-builder/esp-idf/components/mbedtls/mbedtls/library/rsa.c:513
0x40135425: mbedtls_safer_memcmp at /home/kevinh/development/meshtastic/esp32-arduino-lib-builder/esp-idf/components/mbedtls/mbedtls/library/rsa.c:513
  \-> inlined by: mbedtls_rsa_rsassa_pkcs1_v15_sign at /home/kevinh/development/meshtastic/esp32-arduino-lib-builder/esp-idf/components/mbedtls/mbedtls/library/rsa.c:2084
0x401354aa: mbedtls_rsa_rsassa_pss_verify_ext at /home/kevinh/development/meshtastic/esp32-arduino-lib-builder/esp-idf/components/mbedtls/mbedtls/library/rsa.c:513
0x401336a9: rsa_verify_wrap at /home/kevinh/development/meshtastic/esp32-arduino-lib-builder/esp-idf/components/mbedtls/mbedtls/library/pk_wrap.c:88
0x401326dd: mbedtls_pk_sign at /home/kevinh/development/meshtastic/esp32-arduino-lib-builder/esp-idf/components/mbedtls/mbedtls/library/pk.c:493
0x40132716: mbedtls_pk_decrypt at /home/kevinh/development/meshtastic/esp32-arduino-lib-builder/esp-idf/components/mbedtls/mbedtls/library/pk.c:493
0x4013d9d5: mbedtls_x509write_crt_der at /home/kevinh/development/meshtastic/esp32-arduino-lib-builder/esp-idf/components/mbedtls/mbedtls/library/x509write_crt.c:479 (discriminator 2)
0x400eef51: httpsserver::SSLCert::setCert(unsigned char*, unsigned short) at /home/kevinh/development/meshtastic/meshtastic-esp32/.pio/libdeps/tbeam/esp32_https_server/src/SSLCert.cpp:51
  \-> inlined by: cert_write at /home/kevinh/development/meshtastic/meshtastic-esp32/.pio/libdeps/tbeam/esp32_https_server/src/SSLCert.cpp:266
0x400ef18e: std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::~basic_string() at /home/kevinh/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/bits/basic_string.h:544 (discriminator 4)
  \-> inlined by: httpsserver::createSelfSignedCert(httpsserver::SSLCert&, httpsserver::SSLKeySize, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >) at /home/kevinh/development/meshtastic/meshtastic-esp32/.pio/libdeps/tbeam/esp32_https_server/src/SSLCert.cpp:297 (discriminator 4)
0x400de537: taskCreateCert(void*) at /home/kevinh/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/bits/shared_ptr_base.h:925
~/development/meshtastic/meshtastic-esp32$ 
```

log:
```
~/development/meshtastic/Meshtastic-python$ bin/run.sh --noproto
rm: cannot remove 'log_*': No such file or directory
WARNING:root:Not sending packet because protocol use is disabled by noProto
???????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????00:00:00 Emitting reboot packet for serial shell
WARNING:root:Not sending packet because protocol use is disabled by noProto
00:00:00 booted, wake cause 0 (boot count 1), reset_reason=reset
00:00:00 Filesystem files:
00:00:00   /db.proto
00:00:00 I2C device found at address 0x34
00:00:00 axp192 PMU found
00:00:00 I2C device found at address 0x3c
00:00:00 ssd1306 display found
00:00:00 done
00:00:00 Meshtastic swver=1.1.23, hwver=unset
00:00:00 Setting random seed 2381403281
00:00:00 Total heap: 255264
00:00:00 Free heap: 222316
00:00:00 Total PSRAM: 4194252
00:00:00 Free PSRAM: 4194252
00:00:00 NVS: UsedEntries 195, FreeEntries 435, AllEntries 630
00:00:00 Setting default preferences!
00:00:00 Expanding short PSK #1
00:00:00 Installing AES128 key!
00:00:00 Wanted region 0, using Unset
00:00:00 Initial packet id 1038066148, numPacketId 4294967295
00:00:00 Loading saved preferences
00:00:00 Loaded saved preferences version 11
00:00:00 This build does not specify a HW_VERSION
00:00:00 Expanding short PSK #1
00:00:00 Installing AES128 key!
00:00:00 Wanted region 8, using TW
00:00:00 legacy_region=, region=8, NODENUM=0x28979058, dbsize=2
chip id detect 0x3
Detect CHIP :AXP192
OUTPUT Register 0x5f
00:00:00 AXP192 Begin PASS
00:00:00 DCDC1: ENABLE
00:00:00 DCDC2: ENABLE
00:00:00 LDO2: ENABLE
00:00:00 LDO3: ENABLE
[... 154 lines cut ...]
00:01:10 [GPS] Update DB node 0x28979058, rx_time=0
00:01:10 [GPS] Received position from=0x28979058, id=0x593d8c17, payloadlen=0
00:01:10 [GPS] DB update position node=0x28979058 time=0, latI=0, lonI=0
00:01:10 [GPS] Node status update: 2 online, 2 total
00:01:11 [GPS] Plugin position handled=0
00:01:11 [GPS] Enqueuing local (id=0x593d8c17 Fr0x58 To0x58, WantAck0, HopLim3 Portnum=3)
00:01:11 [GPS] Plugin position handled=0
00:01:11 [GPS] Rebroadcasting received floodmsg to neighbors (id=0x593d8c16 Fr0x58 To0xff, WantAck0, HopLim3 Portnum=3 WANTRESP)
00:01:11 [GPS] enqueuing for send (id=0x593d8c16 Fr0x58 To0xff, WantAck0, HopLim2 encrypted)
00:01:11 [GPS] (bw=125, sf=12, cr=4/8) packet symLen=32 ms, payloadSize=22, time 2596 ms
00:01:11 [GPS] txGood=0,rxGood=0,rxBad=0
00:01:11 [GPS] FIXME-update-db Sniffing packet
00:01:11 [GPS] Delivering rx packet (id=0x593d8c16 Fr0x58 To0xff, WantAck0, HopLim3 Portnum=3 WANTRESP)
Trigger powerFSM 3
00:01:11 [GPS] Forwarding to phone (id=0x593d8c16 Fr0x58 To0xff, WantAck0, HopLim3 Portnum=3 WANTRESP)
00:01:11 [GPS] Update DB node 0x28979058, rx_time=0
00:01:11 [GPS] Received position from=0x28979058, id=0x593d8c16, payloadlen=0
00:01:11 [GPS] DB update position node=0x28979058 time=0, latI=0, lonI=0
00:01:11 [GPS] Node status update: 2 online, 2 total
00:01:11 [GPS] Sending response
00:01:11 [GPS] Update DB node 0x28979058, rx_time=0
00:01:11 [GPS] Received position from=0x28979058, id=0x593d8c18, payloadlen=0
00:01:11 [GPS] DB update position node=0x28979058 time=0, latI=0, lonI=0
00:01:11 [GPS] Node status update: 2 online, 2 total
```

mc-hamster, 2020-12-28:
I've tried turning off the timer until after the certs are generated but where ever it's crashing, it's deep in the esp32 sdk. 

All I've been able to track down so far is the key generation routine in the sdk doesn't call yield() so if it generates a 2048 bit key, sometimes takes a while and doesn't give resources back to the chip.

It never crashes when generating a 1024bit key and more frequently crashes when generating a 4096bit key. I would have preferred to use a 1024bit key but browsers are starting to consider 1024bit self signed certs as invalid.

My current path is to find the actual root cause in the sdk and bring in the fix if it has been already addressed up stream.

mc-hamster, 2020-12-31:
Ah ha! 

Fixed it!

I was trying to use a combination of:
disableCore0WDT()
disableCore1WDT()
vTaskDelay()
yield()

but none of those worked. The problem (problem is not the right word, but whatever) is that we're using system tasks which I'm just now getting familiar with. Within a task the proper function to call when you're taking too long is:

esp_task_wdt_reset()

closed 2020-12-31 by mc-hamster