meshtastic/firmware #604: timing going up and down (and not correct)

wakareva, 2020-12-29:
Hi,
on my 2 tbeam I've notice this uncorrect time counting.
As per the video the timing goes up and down and step from minutes to hours and back. In general after 4 minutes it step up to 74 or 93 hours also moving back in time. Plus sometime the time is too much bigger.
I hope the picture/video are usefull.

https://youtu.be/WqUp11axkXk
https://youtu.be/jKp-wvuiO1g
https://youtu.be/aG10zvHD9tM

[image]

mc-hamster, 2020-12-31:
The number of hours on the screen calculates out roughly to some time Jan 1, 1970.

mc-hamster, 2020-12-31:
The node's last_seen is probably being saved before gps lock is made and then after the device gets the gps lock, when we do getTime() it does the math based on the wrong time scale.

wfang2002, 2020-12-31:
I have the same issue and created PR for the fix https://github.com/meshtastic/Meshtastic-device/pull/613

geeksville, 2020-12-31:
thanks!  @wfang2002's fix is exactly correct!  somehow I must have nuked that line accidentally.

commit 96328526 (2020-12-31): Merge pull request #613 from wfang2002/#604_rtc

Shall upgrade RTC currentQuality if higher q available (fix #604)
closed 2020-12-31 by geeksville