meshtastic/firmware #614: SPI transactions in radiolib are super suboptimal

geeksville, 2020-12-31:
while working on pinetab support I noticed that the radio lib spi operations are super inefficent.  Both for regular register operations and bulk packet read/writes.

The fix is to use the more modern two buf implementations of transfer() provided on the nrf52/esp32 (rather than single byte transfers).  This will let the hardware SPI controllers do more of the work.  This would also prevent the linux port from doing lots of extra USB operations/kernel entries.

thebentern, 2022-04-26:
@caveman99 do you have any ideas on this one? I wonder if RadioLib updates might have possibly optimized the SPI transactions since the issue was created.

garthvh, 2022-09-12:
closed in favor of #1374

closed 2022-09-12 by garthvh