meshtastic/firmware #620: t-beam v0.7 power indicate bug

cronyx, 2021-01-02:
hello, I have a t-beam v0.7 and they do not show the battery status, the indicator always shows charging.
[image]

mc-hamster, 2021-01-03:
I may have exposed this bug when fixing #535.

My guess is powerStatus->getHasBattery() is not detecting that the t-beam v0.7 has a battery.

mc-hamster, 2021-01-03:
getHasBattery returns a bool which is set by a class that is inherited from the AXP20X library. That should probably also check the value of the voltage divider in the event the AXP is not present.

There's a lot of inheritance going on here, not something I'd feel comfortable fixing without an v0.7 on hand.

mc-hamster, 2021-01-03:
@cronyx Could you take a picture of the screen that shows the battery percentage and include the log file on this ticket? I think I see how to fix this problem without working in the classes.

cronyx, 2021-01-03:
[image]

mc-hamster, 2021-01-03:
Alright. I think I have an idea of how to fix this.

thank you

thebentern, 2021-02-02:
I think this is the same behavior I'm noticing on the TLora32 T3_V1.6 boards. They always display as if powered. Because of that I believe, they never appear to enter any form of sleep mode, draining my 18650s very quickly.
[image]

mc-hamster, 2021-02-02:
> I think this is the same behavior I'm noticing on the TLora32 T3_V1.6 boards. They always display as if powered. Because of that I believe, they never appear to enter any form of sleep mode, draining my 18650s very quickly.
> [image]

It very well could be the same. Both the 0.7 and your board don't have a power management chip.

thebentern, 2021-02-02:
> It very well could be the same. Both the 0.7 and your board don't have a power management chip.
I'm guessing this one of the reasons the latest iteration of the T-Beams is the preferred hardware. I had hoped to make some very lean usage solar powered relays with these TLora32s, but it seems I may have picked the wrong board for the job. The choice was made in an attempt to save a few bucks and since I didn't have a use for the GPS, given that these would be repeater nodes.

mc-hamster, 2021-02-02:
I think you have picked a good board, but there's a bug somewhere in how we go into a lower power mode.

thebentern, 2021-02-03:
Possibly related to #330 ?

geeksville, 2021-02-14:
yeah - I bet my recent checkin to fix the the voltage sensing fixes this bug.  @thebentern can you see if >=1.1.46 fixes things for you.

thebentern, 2021-02-14:
@geeksville , Thanks! I think we're getting reasonable values now, based on what I'm seeing in the display. 
[image]

I'm a little bit confused though, because my board in router / low power mode still doesn't go into light sleep cycle on battery power (and disconnected from BT). I thought that was controlled by **wait_bluetooth_secs**, but maybe I'm misunderstanding the power state transitions? Any ideas?

```
meshtastic --info
Connected to radio
my_node_num: 2442571748
num_channels: 13
region: "1.0-US"
hw_model: "tlora-v2-1-1.6"
firmware_version: "1.1.46"
packet_id_bits: 32
current_packet_id: 1235425311
node_num_bits: 32
message_timeout_msec: 300000
min_app_version: 20120

preferences {
  wait_bluetooth_secs: 60
  screen_on_secs: 60
  phone_sds_timeout_sec: 4294967295
  mesh_sds_timeout_secs: 4294967295
  ls_secs: 86400
  min_wake_secs: 10
  region: US
  gps_operation: GpsOpMobile
  gps_update_interval: 86400
  gps_attempt_time: 300
  is_router: true
  is_low_power: true
}

```

geeksville, 2021-02-16:
This issue has been mentioned on **Meshtastic**. There might be relevant details there:

https://meshtastic.discourse.group/t/cheep-solar-light-build-progress/2493/8

sachaw, 2022-03-10:
Dis this get resolved?

thebentern, 2022-03-10:
@sachaw it is for my TLoRA v2 1.6 boards and for every board with this type of issue, setting an `adc_multiplier_override` is an option

closed 2022-03-10 by thebentern