meshtastic/firmware #635: Include more detail in the /json/report

ScriptBlock, 2021-01-11:
Please include some additional data in the /json/report endpoint. 

Some ideas that may be beneficial for data collection and reporting are:
- battery charge/status
- time on battery/charge
- environmental details (where possible) - attitude/orientation/acceleration changes of device, ambient temperature
- API or remote IP statistics

ScriptBlock, 2021-01-11:
Ideally this telemetry would be configurable and opt-in (I think @geeksville said it was a future plan to do some of this).

crossan007, 2021-01-11:
I would like this info as well for meshtastic-web

crossan007, 2021-01-11:
Just opened a separate issue for some related items I'd like: https://github.com/meshtastic/Meshtastic-device/issues/636

mc-hamster, 2021-01-17:
@ScriptBlock 

@crossan007 Added power stats.
I added memory and filesystem stats.

Those are the easy stats for us to get. This will go into the next alpha build.

I'm closing this request and marking it done. Please open a new request if there is a specific stat you'd like.

Thanks!

closed 2021-01-17 by mc-hamster