meshtastic/firmware #637: Big Invisible File on SPIFFS

crossan007, 2021-01-12:
I'm not sure how I created this scenario, but I'm documenting here in case it shows up again.

SPIFFS shows as having no files, yet 42168 bytes are used.  The  same usage is reported in the JSON endpoint as on the `/static` page

[image]

```
{
"data": {
"files": [],"filesystem" : {"total" : 414401,"used" : 42168,"free" : 372233}
},
"status": "ok"
}

```

mc-hamster, 2021-01-15:
The hidden file is /db.proto 

I purposely hide it because that file contains sensitive information.

That's also why our files are kept in /static -- to stay away from primary uses of the filesystem.

mc-hamster, 2021-01-15:
The flash is also used to store the public / private key pairs for the web server. That is also hidden from view, but done so at a lower layer than is visible from the spiffs, but it also takes up storage.

closed 2021-01-15 by mc-hamster