meshtastic/firmware #638: Parameter to disable Serial Interface

BoostWillis, 2021-01-13:
The serial interface is implicitly trusted, however repeaters left unattended are vulnerable to physical attacks. More important than device loss, access to the serial interface can compromise entire mesh with a simple `meshtastic --info`, for example.

While I assume the key could still be recovered by delving into binary dumps from esptool.py, that would at least require more time and skill.

Perhaps this could be accomplished by a parameter to disable the serial interface, at least until the next time the device is wiped.

geeksville, 2021-01-15:
also, we should use the same key we will optionally require for changing bluetooth pairing, or to reauthenticate the bluetooth user.

mc-hamster, 2021-03-30:
Set serial_disabled to true and the serial console will be disabled.

closed 2021-03-30 by mc-hamster

geeksville, 2021-04-01:
I think serial_disabled is a good initial effort.  Eventually I'd like to have a bool to "require_admin_keys".  If set, the local device would require that the admin channel (and its secret key) be used for all communication (not just remote).  This would effectively lockdown both the serial and bluetooth interfaces