meshtastic/firmware #643: /json/Report failing

crossan007, 2021-01-17:
Getting a fairly regular stack trace on the latest master branch (84b9028ecb7cef1e9c183a43b8819bc86b1f9a6a)

[HTTPS:I] Request: GET /json/report (FID=57)
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400014fd  PS      : 0x00060230  A0      : 0x801192fc  A1      : 0x3ffd13b0
A2      : 0x003ffd0c  A3      : 0x003ffd08  A4      : 0x000000ff  A5      : 0x0000ff00
A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x00000000  A9      : 0x3ffd1740
A10     : 0x00000000  A11     : 0x000002e8  A12     : 0x3f800000  A13     : 0x00000001
A14     : 0x000000fe  A15     : 0x00000000  SAR     : 0x00000004  EXCCAUSE: 0x0000001c
EXCVADDR: 0x003ffd0c  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff

ELF file SHA256: 0000000000000000

Backtrace: 0x400014fd:0x3ffd13b0 0x401192f9:0x3ffd13c0 0x401220ca:0x3ffd16d0 0x40122106:0x3ffd1760 0x400f8802:0x3ffd17a0 0x400e01d5:0x3ffd1840 0x401d8abb:0x3ffd19b0 0x400df5ee:0x3ffd19d0 0x400df61e:0x3ffd19f0 0x400f159d:0x3ffd1a10 0x400df5ee:0x3ffd1a50 0x400f1316:0x3ffd1a70 0x400f159d:0x3ffd1ad0 0x400df5ee:0x3ffd1b10 0x400f1c25:0x3ffd1b30 0x400eec82:0x3ffd1d00 0x400dea25:0x3ffd1d30 0x400dea4b:0x3ffd1d50 0x400d4e7a:0x3ffd1d70 0x400ed4dd:0x3ffd1d90 0x400d9d38:0x3ffd1dc0 0x400fc71d:0x3ffd1de0

crossan007, 2021-01-17:
stack dump:
```
stack:
0x401192f9: _svfprintf_r at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/vfprintf.c:1437 (discriminator 27)
0x401220ca: __sbprintf at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/vfprintf.c:492
0x40122106: _vsnprintf_r at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/vsnprintf.c:70
0x400f8802: Print::printf(char const*, ...) at C:\Users\Charles\.platformio\packages\framework-arduinoespressif32\cores\esp32/Print.cpp:234
0x400e01d5: handleReport(httpsserver::HTTPRequest*, httpsserver::HTTPResponse*) at c:\users\charles\.platformio\packages\toolchain-xtensa32\xtensa-esp32-elf\include\c++\5.2.0\bits/shared_ptr_base.h:925
0x401d8abb: httpsserver::SSLCert::getCertData() at ??:?
0x400df5ee: fs::File::operator=(fs::File&&) at ??:?
0x400df61e: middlewareSpeedUp240(httpsserver::HTTPRequest*, httpsserver::HTTPResponse*, std::function<void ()>) at c:\users\charles\.platformio\packages\toolchain-xtensa32\xtensa-esp32-elf\include\c++\5.2.0\bits/shared_ptr_base.h:925
0x400f159d: std::function<void ()>::~function() at c:\users\charles\.platformio\packages\toolchain-xtensa32\xtensa-esp32-elf\include\c++\5.2.0/functional:1974
  \-> inlined by: std::_Head_base<2u, std::function<void ()>, false>::~_Head_base() at c:\users\charles\.platformio\packages\toolchain-xtensa32\xtensa-esp32-elf\include\c++\5.2.0/tuple:102 
  \-> inlined by: std::_Tuple_impl<2u, std::function<void ()> >::~_Tuple_impl() at c:\users\charles\.platformio\packages\toolchain-xtensa32\xtensa-esp32-elf\include\c++\5.2.0/tuple:339     
  \-> inlined by: std::_Tuple_impl<1u, httpsserver::HTTPResponse*, std::function<void ()> >::~_Tuple_impl() at c:\users\charles\.platformio\packages\toolchain-xtensa32\xtensa-esp32-elf\include\c++\5.2.0/tuple:180
  \-> inlined by: std::_Tuple_impl<0u, httpsserver::HTTPRequest*, httpsserver::HTTPResponse*, std::function<void ()> >::~_Tuple_impl() at c:\users\charles\.platformio\packages\toolchain-xtensa32\xtensa-esp32-elf\include\c++\5.2.0/tuple:180
  \-> inlined by: std::tuple<httpsserver::HTTPRequest*, httpsserver::HTTPResponse*, std::function<void ()> >::~tuple() at c:\users\charles\.platformio\packages\toolchain-xtensa32\xtensa-esp32-elf\include\c++\5.2.0/tuple:463
  \-> inlined by: std::_Bind<void (*(httpsserver::HTTPRequest*, httpsserver::HTTPResponse*, std::function<void ()>))(httpsserver::HTTPRequest*, httpsserver::HTTPResponse*, std::function<void ()>)>::~_Bind() at c:\users\charles\.platformio\packages\toolchain-xtensa32\xtensa-esp32-elf\include\c++\5.2.0/functional:1058
  \-> inlined by: std::_Function_base::_Base_manager<std::_Bind<void (*(httpsserver::HTTPRequest*, httpsserver::HTTPResponse*, std::function<void ()>))(httpsserver::HTTPRequest*, httpsserver::HTTPResponse*, std::function<void ()>)> >::_M_destroy(std::_Any_data&, std::integral_constant<bool, false>) at c:\users\charles\.platformio\packages\toolchain-xtensa32\xtensa-esp32-elf\include\c++\5.2.0/functional:1726
  \-> inlined by: std::_Function_base::_Base_manager<std::_Bind<void (*(httpsserver::HTTPRequest*, httpsserver::HTTPResponse*, std::function<void ()>))(httpsserver::HTTPRequest*, httpsserver::HTTPResponse*, std::function<void ()>)> >::_M_manager(std::_Any_data&, std::_Any_data const&, std::_Manager_operation) at c:\users\charles\.platformio\packages\toolchain-xtensa32\xtensa-esp32-elf\include\c++\5.2.0/functional:1750
0x400df5ee: fs::File::operator=(fs::File&&) at ??:?
0x400f1316: httpsserver::validationMiddleware(httpsserver::HTTPRequest*, httpsserver::HTTPResponse*, std::function<void ()>) at C:\Users\Charles\Documents\GitHub\Meshtastic\Meshtastic-device/.pio\libdeps\tbeam\esp32_https_server\src/HTTPConnection.cpp:631
0x400f159d: std::function<void ()>::~function() at c:\users\charles\.platformio\packages\toolchain-xtensa32\xtensa-esp32-elf\include\c++\5.2.0/functional:1974
  \-> inlined by: std::_Head_base<2u, std::function<void ()>, false>::~_Head_base() at c:\users\charles\.platformio\packages\toolchain-xtensa32\xtensa-esp32-elf\include\c++\5.2.0/tuple:102 
  \-> inlined by: std::_Tuple_impl<2u, std::function<void ()> >::~_Tuple_impl() at c:\users\charles\.platformio\packages\toolchain-xtensa32\xtensa-esp32-elf\include\c++\5.2.0/tuple:339     
  \-> inlined by: std::_Tuple_impl<1u, httpsserver::HTTPResponse*, std::function<void ()> >::~_Tuple_impl() at c:\users\charles\.platformio\packages\toolchain-xtensa32\xtensa-esp32-elf\include\c++\5.2.0/tuple:180
  \-> inlined by: std::_Tuple_impl<0u, httpsserver::HTTPRequest*, httpsserver::HTTPResponse*, std::function<void ()> >::~_Tuple_impl() at c:\users\charles\.platformio\packages\toolchain-xtensa32\xtensa-esp32-elf\include\c++\5.2.0/tuple:180
  \-> inlined by: std::tuple<httpsserver::HTTPRequest*, httpsserver::HTTPResponse*, std::function<void ()> >::~tuple() at c:\users\charles\.platformio\packages\toolchain-xtensa32\xtensa-esp32-elf\include\c++\5.2.0/tuple:463
  \-> inlined by: std::_Bind<void (*(httpsserver::HTTPRequest*, httpsserver::HTTPResponse*, std::function<void ()>))(httpsserver::HTTPRequest*, httpsserver::HTTPResponse*, std::function<void ()>)>::~_Bind() at c:\users\charles\.platformio\packages\toolchain-xtensa32\xtensa-esp32-elf\include\c++\5.2.0/functional:1058
[... cut ...]
  \-> inlined by: _GLOBAL__sub_I__ZN11concurrency20NotifiedWorkerThread6notifyEjb at C:\Users\Charles\Documents\GitHub\Meshtastic\Meshtastic-device/src\concurrency/NotifiedWorkerThread.cpp:85
0x400ed4dd: ThreadController::runOrDelay() at C:\Users\Charles\Documents\GitHub\Meshtastic\Meshtastic-device/.pio\libdeps\tbeam\Thread/ThreadController.cpp:153
0x400d9d38: loop() at C:\Users\Charles\Documents\GitHub\Meshtastic\Meshtastic-device/src/main.cpp:601
0x400fc71d: uartDetectBaudrate at C:\Users\Charles\.platformio\packages\framework-arduinoespressif32\cores\esp32/esp32-hal-uart.c:606
```

crossan007, 2021-01-17:
It looks like these lines are causing the panic
```
        //res->printf("\"psram_total\": %s,\n", ESP.getPsramSize());
        //res->printf("\"psram_free\": %s,\n", ESP.getFreePsram());
```

crossan007, 2021-01-17:
This might be related https://github.com/espressif/arduino-esp32/issues/1774

mc-hamster, 2021-01-17:
fixed

closed 2021-01-17 by mc-hamster