meshtastic/firmware #647: Unable to connect device to main Wifi AP

rodavrow, 2021-01-18:
I am trying to connect one of the devices to my main AP in the house, with an external antenna outside as a sort of basestation.
I cannot connect it to the Wifi though, I’m using the meshtastic python script to set the parameters like this:

meshtastic --set wifi_ap_mode false --setstr wifi_ssid Meshtastic_NET --setstr wifi_password xxxxxxxxxxxxxxxxxx

When I check the info with ‘meshtastic --info’ everything come back fine, but on the device status screen it still shows the SoftAP MeshtasticAdmin and it’s details.

The hardware I am using is the TTGO T3 V1.6-20180606 boards.
Firmware is latest - firmware-tlora-v2-1-1.6-EU865-1.1.31.bin

I have tried:

Reflashing different boards several times using the command line
Changing the details of the SoftAP, --info shows correctly on the command line but it does not change on the device status screen or what it actually broadcasts
Rebooting - changes persist when checking with --info but the screen and reality do not

geeksville, 2021-01-18:
This issue has been mentioned on **Meshtastic**. There might be relevant details there:

https://meshtastic.discourse.group/t/unable-to-connect-device-to-main-wifi-ap/2191/6

rodavrow, 2021-01-18:
Yes that was me :) mc_hamster asked me to open the issue here.

mc-hamster, 2021-01-18:
Schematic 

https://github.com/Xinyuan-LilyGO/LilyGO-T-Beam/blob/master/schematic/T3_V1.6.pdf

rodavrow, 2021-01-18:
Tested:

V1.1.20 - Works as expected, able to configure Wifi and it connects to main AP. Also device SoftAP does not enable by deafult.
V1.1.23 > onwards - Functionality lost, unable to change Wifi mode or settings.

In either case, adding a button to Pin 12 allows usage (as users are doing)

rodavrow, 2021-01-18:
https://github.com/meshtastic/Meshtastic-device/compare/1.1.20...1.1.23

mc-hamster, 2021-01-18:
Fixed in next build.

closed 2021-01-18 by mc-hamster

geeksville, 2021-01-26:
This issue has been mentioned on **Meshtastic**. There might be relevant details there:

https://meshtastic.discourse.group/t/new-device-alpha-1-1-33/2270/1

OpenSource-For-Freedom, 2023-08-27:
A similar issue connecting TBeam (v2.2.3) to LAN, then disassociates. Saw an NTP error, so changed NTP to NIST, then saw DNS error, and used local DNS as with my other TBeam. Here are the Logs: 

DEBUG | 05:11:18 149 [Power] Battery: usbPower=1, isCharging=1, batMv=4166, batPct=100
INFO  | 05:11:18 149 [Power] USB plugged In
INFO  | 05:11:28 159 [mqtt] Using non-TLS-encrypted session
INFO  | 05:11:28 159 [mqtt] Attempting to connect directly to MQTT server mqtt.meshtastic.org, port: 1883, username: meshdev, password: large4cats
[173368][E][WiFiGeneric.cpp:1578] hostByName(): DNS Failed for mqtt.meshtastic.org
ERROR | 05:11:42 173 [mqtt] Failed to contact MQTT server directly (5/5)...
DEBUG | 05:11:42 173 [Power] Battery: usbPower=1, isCharging=1, batMv=4166, batPct=100
INFO  | 05:11:4[173478][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
[173478][W][WiFiGeneric.cpp:1057] _eventCallback(): Reason: 8 - ASSOC_LEAVE
2 173 [WifiConnect] Reconnecting to WiFi access point ATT
WARN  | 05:11:43 173 [WifiConnect] WiFi lost connection. Reason: 8
[179247][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 4 - STA_CONNECTED
DEBUG | 05:11:48 179 WiFi-Event 4: Connected to access point
[179249][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 7 - STA_GOT_IP
[179256][D][WiFiGeneric.cpp:1098] _eventCallback(): STA IP: 192.168.1.222, MASK: 255.255.255.0, GW: 192.168.1.254
DEBUG | 05:11:48 179 WiFi-Event 7: Obtained IP address: Using non-TLS-encrypted session
INFO  | 05:11:48 179 Attempting to connect directly to MQTT server mqtt.meshtastic.org, port: 1883, username: meshdev, password: large4cats
INFO  | 05:11:49 179 [WifiConnect] Reconnecting to WiFi access point ATT
DEBUG | 05:11:54 184 [WifiConnect] Updating NTP time from time-a-g.nist.gov
[193292][E][WiFiGeneric.cpp:1578] hostByName(): DNS Failed for mqtt.meshtastic.org
ERROR | 05:12:02 193 [WifiConnect] Failed to contact MQTT server directly (1/5)...
[193304][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
[193304][W][WiFiGeneric.cpp:1057] _eventCallback(): Reason: 8 - ASSOC_LEAVE
DEBUG | 05:12:02 193 [WifiConnect] WiFi-Event 5: Disconnected from WiFi access point
WARN  | 05:12:02 193 [WifiConnect] WiFi lost connection. Reason: 8
[198292][E][WiFiUdp.cpp:172] beginPacket(): could not get host from dns: 11
[198292][E][WiFiUdp.cpp:185] endPacket(): could not send data: 118
DEBUG | 05:12:08 199 [WifiConnect] NTP Update failed
DEBUG | 05:12:08 199 [Power] Battery: usbPower=1, isCharging=1, batMv=4167, batPct=100
INFO  | 05:12:09 200 [WifiConnect] Reconnecting to WiFi access point ATT
[205986][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 4 - STA_CONNECTED
DEBUG | 05:12:15 205 WiFi-Event 4: Connected to access point
[205988][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 7 - STA_GOT_IP
[205995][D][WiFiGeneric.cpp:1098] _eventCallback(): STA IP: 192.168.1.222, MASK: 255.255.255.0, GW: 192.168.1.254
DEBUG | 05:12:15 206 WiFi-Event 7: Obtained IP address: Using non-TLS-encrypted session
INFO  | 05:12:15 206 Attempting to connect directly to MQTT server mqtt.meshtastic.org, port: 1883, username: meshdev, password: large4cats
DEBUG | 05:12:15 206 [WifiConnect] Updating NTP time from time-a-g.nist.gov
[220031][E][WiFiGeneric.cpp:1578] hostByName(): DNS Failed for mqtt.meshtastic.org
ERROR | 05:12:29 220 [WifiConnect] Failed to contact MQTT server directly (2/5)...
[220032][E][WiFiUdp.cpp:172] beginPacket(): could not get host from dns: 11
DEBUG | 05:12:30 221 [WifiConnect] NTP Update failed
DEBUG | 05:12:30 221 [Power] Battery: usbPower=1, isCharging=1, batMv=4167, batPct=100
INFO  | 05:12:30 221 [mqtt] Using non-TLS-encrypted session
INFO  | 05:12:30 221 [mqtt] Attempting to connect directly to MQTT server mqtt.meshtastic.org, port: 1883, username: meshdev, password: large4cats
[235158][E][WiFiGeneric.cpp:1578] hostByName(): DNS Failed for mqtt.meshtastic.org
ERROR | 05:12:44 235 [mqtt] Failed to contact MQTT server directly (3/5)...
DEBUG | 05:12:50 241 [Power] Battery: usbPower=1, isCharging=1, batMv=4167, batPct=100
DEBUG | 05:13:10 261 [Power] Battery: usbPower=1, isCharging=1, batMv=4169, batPct=100
INFO  | 05:13:14 265 [mqtt] Using non-TLS-encrypted session
INFO  | 05:13:14 265 [mqtt] Attempting to connect directly to MQTT server mqtt.meshtastic.org, port: 1883, username: meshdev, password: large4cats
[279185][E][WiFiGeneric.cpp:1578] hostByName(): DNS Failed for mqtt.meshtastic.org
ERROR | 05:13:28 279 [mqtt] Failed to contact MQTT server directly (4/5)...
DEBUG | 05:13:30 281 [Power] Battery: usbPower=1, isCharging=1, batMv=4169, batPct=100
DEBUG | 05:13:50 301 [Power] Battery: usbPower=1, isCharging=1, batMv=4169, batPct=100
INFO  | 05:13:58 309 [mqtt] Using non-TLS-encrypted session
INFO  | 05:13:58 309 [mqtt] Attempting to connect directly to MQTT server mqtt.meshtastic.org, port: 1883, username: meshdev, password: large4cats
[323212][E][WiFiGeneric.cpp:1578] hostByName(): DNS Failed for mqtt.meshtastic.org
ERROR | 05:14:12 323 [mqtt] Failed to contact MQTT server directly (5/5)...
DEBUG | 05:14:12 323 [Power] Battery: usbPower=1, isCharging=1, batMv=4170, batPct=100
INFO  | 05:[323384][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
[323384][W][WiFiGeneric.cpp:1057] _eventCallback(): Reason: 8 - ASSOC_LEAVE
14:12 323 [WifiConnect] Reconnecting to WiFi access point ATT
[... cut ...]
DEBUG | 05:18:10 560 [Power] Battery: usbPower=1, isCharging=1, batMv=4174, batPct=100
INFO  | 05:18:14 564 [mqtt] Using non-TLS-encrypted session
INFO  | 05:18:14 564 [mqtt] Attempting to connect directly to MQTT server mqtt.meshtastic.org, port: 1883, username: meshdev, password: large4cats
[578969][E][WiFiGeneric.cpp:1578] hostByName(): DNS Failed for mqtt.meshtastic.org
ERROR | 05:18:28 578 [mqtt] Failed to contact MQTT server directly (4/5)...
DEBUG | 05:18:30 580 [Power] Battery: usbPower=1, isCharging=1, batMv=4174, batPct=100