meshtastic/firmware #657: Ability to change the bluetooth name

lennisthemenace, 2021-01-23:
Can you please add the ability to change the Bluetooth name of a device. Either in the app or python api?

geeksville, 2021-02-17:
alas - some old versions of android have a bug so that the filtering to tell us what BLE devices are visible are busted.  The class based (UUID) filtering we'd like to use is busted, so for those phones we have to use a rule like "the name starts with Meshtastic".  Which is my long way of saying we probably won't get to this (good) idea for a while... ;-)

lennisthemenace, 2021-02-18:
Would it then be possible to rename in the api but it automatically has meshtastic preappended?

geeksville, 2021-02-19:
sure!

garthvh, 2022-01-22:
Can we look at this again, this is a confusing spot for many users who are trying to customize their devices.  Do we know what android version is unable to filter UUID's

andrekir, 2022-01-23:
> Do we know what android version is unable to filter UUID's

current min API  21+ supports UUID filtering.

garthvh, 2022-01-24:
> > Do we know what android version is unable to filter UUID's
> 
> current min API 21+ supports UUID filtering.

And we currently recommend Android 6+ right?

andrekir, 2022-01-25:
> And we currently recommend Android 6+ right?

correct. minimum API level 21 is Android 5.0 (Lollipop) but Android 6+ (Marshmallow) is recommended.

garthvh, 2022-01-27:
So about 2% of Android usage would be lost to fix this, privacy updates over time may make it more urgent.

LolloDev5123, 2022-01-31:
Also change hotspot name? From Meshtastic 0f0f to some other unique code, I like the project, but my mesh is so private that I don't want people to search the name of a wifi they see and reveal that they are near a mesh node, also, there should be an option for the AP to become a private/un discoverable wifi net?

mc-hamster, 2022-01-31:
Please open a separate feature request for that. Good idea

garthvh, 2022-06-14:
Closing this, other pulls in progress