meshtastic/firmware #662: Improve pointer / bearing accuracy

android606, 2021-01-27:
At walking speeds and relatively close distances, a GPS error of 3-5m+ can make your position pretty uncertain, which makes bearing calculations  and therefore make the pointer arrow on the screen noticeably inaccurate.

We should look into ways to make bearing calculations that are more accurate. Is it possible to use more points to generate an "average" bearing, perhaps?

See discussion at: [https://meshtastic.discourse.group/t/direction-of-compass-orientation-device/1282/14](discourse)

geeksville, 2021-01-27:
This issue has been mentioned on **Meshtastic**. There might be relevant details there:

https://meshtastic.discourse.group/t/direction-of-compass-orientation-device/1282/17

mc-hamster, 2021-12-25:
closing this issue.

averaging really may not work since the drift will be the same for the averaged points.

closed 2021-12-25 by mc-hamster