meshtastic/firmware #668: Store & Request Plugin

mc-hamster, 2021-01-29:
New plugin that would **store** received packets on a router node.

A node can then send a **request** to the mesh to be the messages that it may have missed. The request can be automatic based on a set metric or manually triggered.

A router would send a periodic heartbeat to let connected devices know that it is on the mesh.

Need to support n-number of routers, but will start with just one. Will use the port "PortNum_STORE_REQUEST_APP".

sachaw, 2021-01-29:
This should be able to be disabled when the configuring the node, maybe a nested configuration i.e routerSettings

mc-hamster, 2021-01-29:
For now, it can go to the same place as the other plugin configurations in RadioConfig.UserPreferences but @geeksville discussed making a separate name space just for plugins.

closed 2021-12-12 by sachaw