meshtastic/firmware #670: Feature idea: Notify when no updates are received from mesh user after timeout

android606, 2021-01-29:
This feature idea is relevant to folks using this as a hiking group or kid-tracking safety device.

The idea is that, as a mesh user, you could apply a "watch" on another mesh user. This would enable these things: 
1. GPS points received from that user would be stored.
2. If no packets are received from that user within a predetermined period of time (5 minutes? 10 minutes?) then an alert is placed on the screen with a pointer pointing at their last known location.
3. An alert should probably also be set in the app, and the GPS points stored for the user could be used to show their last known track, to more easily travel the same path they travelled and therefore find them.

That way, if your kids drop the device down a crevasse or into the ocean or something, you still know where to go to find them.

These alerts might also be activated in the case where the user presses their emergency "I need help" button mentioned in #60

mc-hamster, 2021-02-23:
If anyone one picks up this request, #2 from the description above is implemented in the store and forward plug-in.

sachaw, 2023-01-04:
This can be implemented client side, I have added last heard counters in the webui, if you could express how exactly you would want this to look/function, we could look to add it to the apps.