meshtastic/firmware #672: Use index based word encoding to save bandwidth

henricazottes, 2021-02-01:
Hi,

I'd like to propose a feature aiming to reduce bandwidth needed to send text. Have you already though about something similar?

As the mesh network is for transport only and thus data agnostic, we could use the smartphone app to perform encoding before sending and decoding on receive.

For example, with little search I found that with index based encoding we could reach 40% reduction on French texts:
- [an-array-of-french-words](https://github.com/words/an-array-of-french-words) features 336K+ words
- Log²(336k) < 19 bits→we only need 2.4 bytes to index any french word (I think we could even reduce the dictionary to get under 262K words and use 18 bits only).
- French words have an average of 4.8 characters length
- UTF-8 chars are 1 byte long, so average of 4.8 bytes for a french word versus 2.4 bytes when indexed. Thus 50% reduction.
- GSM-7 (used for sms) chars are 7 bits long, so average of 4.2 bytes for a french word versus 2.4 bytes when indexed. Thus 40% reduction.

The same apply for English ([an-array-of-english-words](https://github.com/words/an-array-of-english-words) 275K+ words, 4.7 chars long words on average).

Obviously this technique won't work for numbers, it means we need to be able to mix indexed and non indexed elements in the payload.

Anything I'm missing which would make this not worth it?

Average sources: https://ans.wiki/q/quelle-est-la-longueur-moyenne-des-mots-du-dictionnaire/

crossan007, 2021-02-04:
I would think the space used by the index would exceed the available space on the device

sachaw, 2021-02-04:
I think a better implementation would be to setup a custom app that handles this, not a native feature, It wouldn't be hard to do with any any of the existing libraries.

henricazottes, 2021-02-04:
Of course I was talking about a companion app feature. I didn't check the other projects, this issue totally belongs to Meshtastic-Android...