meshtastic/firmware #673: Typo on Website

karlhorky, 2021-02-01:
Hi, thanks for this project, really cool!

**Describe the bug**

The Meshtastic website at https://meshtastic.letstalkthis.com/ has a typo on it (`Andoid` without an `r`)

I looked for the website source code under [the `meshtastic` organization](https://github.com/meshtastic) to do a pull request but I guess it's not open source?

**To Reproduce**
Steps to reproduce the behavior:
1. Go to https://meshtastic.letstalkthis.com/
2. Scroll down
3. See typo

**Expected behavior**

No Typo

**Screenshots**

[image]

**Device info:**
 - Device model: -
 - Software Version -

**Smartphone information (if relevant):**
 - Device: -
 - OS: -
 - App Version -

geeksville, 2021-02-14:
thanks for the note.  alas, we didn't make that (helpful) webpage.  I'll post where the author might see this.

geeksville, 2021-02-14:
This issue has been mentioned on **Meshtastic**. There might be relevant details there:

https://meshtastic.discourse.group/t/beginners-guide/717/10

closed 2021-02-14 by geeksville

rezl, 2021-04-12:
This has been fixed. :)