meshtastic/firmware #675: Open-Source Hardware

arthurwolf, 2021-02-03:
Hello.

Is there any hardware this can be run on, that is Open-Source Hardware (https://www.oshwa.org/)?

I'd like to make my custom system based on this, but if there is an Open-Source platform this can be run on, I'd much rather contribute/participate in existing efforts than start my own from scratch.

Even just partial hardware ( boards that would require me to design daughterboards/interface with them ) would be helpful.

I'm asking mostly since the project is Open-Source, so that spirit seems to be there (thumbs up!) but I couldn't find something similar when doing research on the hardware side of things.

Thanks a lot to anyone who has any information on this.
Best regards.

f-asa, 2021-02-03:
its not on oshwa but check this out, https://github.com/BigCorvus/LoRa-BLE-Relay-v2

f-asa, 2021-02-03:
if you want a quick win starter pack, use a Ebyte E22 module an some ESP32 or NRF chip

arthurwolf, 2021-02-03:
Thank you so much, that's exactly what I was looking for.
What's the status of the project, is it confirmed to work, has it been
tested by many? (I'm going to read the forum and all the info I can find,
you don't need to answer if it's already explained there).

arthurwolf, 2021-02-03:
Oh, it's not OSHWA, just noticed. I see no license at all, is there a plan to add one, and if so what will it be?

geeksville, 2021-02-14:
also - check the forum from posts by BigCorvus.  He's made a nice open source hardware design he has up on github.

(for now - I'm closing this bug - because you'll probably get more useful answers with a forum post on meshtastic.discourse.group).  G

closed 2021-02-14 by geeksville