meshtastic/firmware #677: Device reboots everytime I sent a message using the meshtastic app

mofosyne, 2021-02-06:
**Describe the bug**
Reboots when I try to send a message via the meshtastic app.

**To Reproduce**

* Installed firmware-tbeam-ANZ-1.1.33.bin to a ttgo tbeam v1.1
* I then installed meshtastic
* Paired it to the app
* Saw it recommends updating
* Updated
* Then tried to send a message
* Saw via serial console that the device rebooted immediately afterwards
* Tried multiple times to send a message and saw repeated reboots

**Expected behavior**
Should not reboot but instead should have just sent a message

**Device info:**
 - Device model: ttgo tbeam v1.1
 - Software Version v1.1.33
 - Did not install oled so had to pair via usb serial to see the bluetooth pin code (On a side note, the main documentation was confusing on how to pair.)

**Additional context**

Serial dump:

```
                                                     Trigger powerFSM 9
                                                                       Trigger powerFSM 9
                                                                                         Trigger powerFSM 9
                                                                                                           Trigger powerFSM 9
                                                                                                                             Trigger powerFSM 9
          Trigger powerFSM 9
                            Trigger powerFSM 9
                                              Trigger powerFSM 9
                                                                Trigger powerFSM 9
                                                                                  Trigger powerFSM 9
                                                                                                    Trigger powerFSM 9
                                                                                                                      Trigger powerFSM 9

<!------- Was doing OTA update here ----->

   Trigger powerFSM 9
                     Trigger powerFSM 9
                                       Trigger powerFSM 9
                                                         Trigger powerFSM 9
                                                                           Trigger powerFSM 9
                                                                                             Trigger powerFSM 9
                                                                                                               Trigger powerFSM 9
                                                                                                                                 Trigger powerFSM 9
              Trigger powerFSM 9
                                Trigger powerFSM 9
                                                  Trigger powerFSM 9
                                                                    Trigger powerFSM 9
                                                                                      Trigger powerFSM 9
                                                                                                        00:15:02 [GPS] WANT GPS=0
00:15:02 [GPS] publishing GPS lock=0
00:15:02 [GPS] WARNING: Using fixed position
00:15:02 [GPS] got gps notify time=0, lat=0, bat=0
00:15:02 [GPS] DB update position node=0x50d5de0 time=0, latI=0, lonI=0
00:15:02 [GPS] Node status update: 0 online, 1 total
00:15:02 [GPS] Sending position to mesh (wantReplies=1)
00:15:02 [GPS] Update DB node 0x50d5de0, rx_time=0
00:15:02 [GPS] Received position from=0x50d5de0, id=0x4f33b238, payloadlen=0
00:15:02 [GPS] DB update position node=0x50d5de0 time=0, latI=0, lonI=0
00:15:02 [GPS] Node status update: 0 online, 1 total
00:15:02 [GPS] Sending response
00:15:02 [GPS] Update DB node 0x50d5de0, rx_time=0
00:15:02 [GPS] Received position from=0x50d5de0, id=0x4f33b239, payloadlen=0
00:15:02 [GPS] DB update position node=0x50d5de0 time=0, latI=0, lonI=0
[... 662 lines cut ...]
00:24:48 Filesystem files:
00:24:48   /db.proto
00:24:48 No I2C devices found
00:24:48 Meshtastic swver=1.1.34, hwver=1.0
00:24:48 Setting random seed 4029039779
00:24:48 Total heap: 254892
00:24:48 Free heap: 222680
00:24:48 Total PSRAM: 4194252
00:24:48 Free PSRAM: 4194252
00:24:48 NVS: UsedEntries 88, FreeEntries 542, AllEntries 630
00:24:48 Setting default preferences!
00:24:48 Expanding short PSK #1
00:24:48 Installing AES128 key!
00:24:48 Wanted region 0, using Unset
00:24:48 Initial packet id 244248018, numPacketId 4294967295
00:24:48 Loading saved preferences
00:24:48 Loaded saved preferences version 11
00:24:48 Expanding short PSK #1
00:24:48 Installing AES128 key!
00:24:48 Wanted region 6, using ANZ
00:24:48 legacy_region=1.0, region=6, NODENUM=0x50d5de0, dbsize=1
00:24:48 Read RTC time as 1488 (cur millis 90) quality=0
00:24:48 WANT GPS=1
00:24:48 Setting GPS power=1
```

geeksville, 2021-02-07:
@mofosyne thank you for the great log.  That let me definitely find the cause of the reboot.  It was actually two things:

## 1: we didn't find the radio when we rebooted

```
00:23:59 RF95 init result -2
00:23:59 Warning: Failed to find RF95 radio
00:23:59 NOTE! Recording critical error 3, address=0
```

## 2: but then we (erroniously) rebooted because of it

```
assertion "iface" failed: file "src/mesh/ReliableRouter.h", line 133, function: void ReliableRouter::setNextTx(PendingPacket*)
abort() was called at PC 0x4010995b on core 0
```

I'll fix the second problem right now.  But for the first problem: that's really puzzling.  Did manually reseting the board fix the problem or does it still continue to not find the radio chip?  Prior to the firmware update did it happily find the radio chip and send messages with it?

mofosyne, 2021-02-07:
Pretty constant loss of modem. It's a tbeam v1.1 .

Maybe there should be a hardware bootup test to help check what exist. E.g. notify user if they possibly loaded the wrong firmware perhaps?

geeksville, 2021-02-07:
heh!  there actually is: Recording critical error 3, address=0
But the android app doesn't show that yet in the GUI
It does put a critical fault screen up on the OLED but without the oled you wouldn't get that info :-(

In a few minutes (after I fix the assertion) I'll look more carefully at your log to figure out what's up with that.  Can you add a photo of this board? (I want to check the radio type)

geeksville, 2021-02-07:
hmm - also can you hold your finger on the board while it is hooked to USB power and feel if the axp192 is hot?  that chip is near the USB port.

I also see and unexpected "ERROR: No UBLOX GPS found"

geeksville, 2021-02-07:
also, can you double check that you originally flashed "firmware-tbeam-ANZ-1.1.33.bin"?  Based on the logs I think you might have accidentally flashed firmware-tbeam0.7-ANZ-1.1.33.bin instead.

geeksville, 2021-02-07:
reboot fixed in https://github.com/meshtastic/Meshtastic-device/pull/678/commits/486b03e985f56eb68a60f9b9b53add7c510a4100

closed 2021-02-07 by geeksville

mofosyne, 2021-02-07:
I think you might be correct. I've reflash again. Also this time found that I can press the middle button to disable bluetooth password for a default pass of `123456`.

Below is my connection after pressing reboot then sending a hello message via my meshtastic node.

```
▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒                                                        ▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒                                                        ▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒                                                        ▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒                                                        ▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒                                                        ▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒                                                        ▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒00:00:00 Emitting reboot pack                                                        et for serial shell
▒▒H00:00:00 booted, wake cause 0 (boot count 1), reset_reason=reset
00:00:00 Filesystem files:
00:00:00   /db.proto
00:00:00 I2C device found at address 0x34
00:00:00 axp192 PMU found
00:00:00 done
00:00:00 Meshtastic swver=1.1.33, hwver=1.0-ANZ
00:00:00 Setting random seed 2365151295
00:00:00 Total heap: 254316
00:00:00 Free heap: 222000
00:00:00 Total PSRAM: 4194252
00:00:00 Free PSRAM: 4194252
00:00:00 NVS: UsedEntries 88, FreeEntries 542, AllEntries 630
00:00:00 Setting default preferences!
00:00:00 Expanding short PSK #1
00:00:00 Installing AES128 key!
00:00:00 Wanted region 0, using Unset
00:00:00 Initial packet id 1656814609, numPacketId 4294967295
00:00:00 Loading saved preferences
00:00:00 Loaded saved preferences version 11
00:00:00 Expanding short PSK #1
00:00:00 Installing AES128 key!
00:00:00 Wanted region 6, using ANZ
00:00:00 legacy_region=1.0-ANZ, region=6, NODENUM=0x50d5de0, dbsize=1
chip id detect 0x3
                  Detect CHIP :AXP192
                                     OUTPUT Register 0x5f
                                                         00:00:00 AXP192 Begin P                                                        ASS
00:00:00 DCDC1: ENABLE
00:00:00 DCDC2: ENABLE
00:00:00 LDO2: ENABLE
00:00:00 LDO3: ENABLE
00:00:00 DCDC3: ENABLE
00:00:00 Exten: ENABLE
00:00:00 ----------------------------------------
00:00:00 DCDC1: ENABLE
00:00:00 DCDC2: ENABLE
00:00:00 LDO2: ENABLE
00:00:00 LDO3: ENABLE
00:00:00 DCDC3: ENABLE
00:00:00 Exten: ENABLE
SRC REG:0xc4
            Charging enable is enable
                                     Charging target-voltage : 0x2
                                                                   end when the                                                         charge current is lower than 10% of the set value
                                                 Charge current : 700.00 mA
                                                                           00:00                                                        :00 Battery: usbPower=1, isCharging=1, batMv=3533, batPct=5
00:00:00 Read RTC time as 0 (cur millis 114) quality=0
00:00:00 WANT GPS=1
00:00:00 Setting GPS power=1
00:00:00 Connected to UBLOX GPS successfully
00:00:01 Starting meshradio init...
00:00:01 (bw=125, sf=12, cr=4/8) packet symLen=32 ms, payloadSize=16, time 2269                                                         ms
00:00:01 Set radio: name=LongSlow, config=3, ch=6, power=0
00:00:01 Radio myRegion->freq: 916.000000
00:00:01 Radio myRegion->spacing: 0.500000
00:00:01 Radio myRegion->numChannels: 20
00:00:01 Radio channel_num: 6
00:00:01 Radio frequency: 919.000000
00:00:01 Short packet time: 2269 msec
00:00:01 Set radio: final power level=17
00:00:01 RF95 init result -2
00:00:01 Warning: Failed to find RF95 radio
00:00:01 Starting meshradio init...
[... 272 lines cut ...]
06:16:47 FIXME-update-db Sniffing packet
06:16:47 Delivering rx packet (id=0xb07a448e Fr0xe0 To0xff, WantAck1, HopLim0 Portnum=1 rxtime=1612678607)
Trigger powerFSM 3
                  06:16:47 Forwarding to phone (id=0xb07a448e Fr0xe0 To0xff, WantAck1, HopLim0 Portnum=1 rxtime=1612678607)
06:16:47 Update DB node 0x50d5de0, rx_time=1612678607
06:16:47 Received text msg from=0x50d5de0, id=-1334164338, msg=hello
Trigger powerFSM 5
                  06:16:47 Plugin text handled=0
06:16:47 Adding packet record (id=0xb07a448e Fr0xe0 To0xff, WantAck1, H06:opL16:47 im1( PoPortrtnunum=1 yet m=1int rxerested in packets)
time=161206:16:678647 07)Telling client we haeve new packets 7
06:16:47 BLE notify fr06:omNum
g6:47 enque06:16:uing47  BLE reading a uint32
  a uint32
=0xb07a448e Fr0xe0 To0xff, WantAck0, HopLim1 encrypted rxtime=1612678607)
06:16:47 (bw=125, sf=12, cr=4/8) packet symLen=32 ms, payloadSize=27, time 2859 ms
06:16:47 txGood=4,rxGood=0,rxBad=0
06:16:47 AirTime - Packet transmitted : 2859ms
06:16:47 getFromRadio, state=6
06:16:47 phone downloaded packet (id=0xb07a448e Fr0xe0 To0xff, WantAck1, HopLim0 Portnum=1 rxtime=1612678607)
06:16:47 encoding toPhone packet to phone variant=2, 39 bytes
06:16:47 BLE fromRadio called omlen=1, ourlen=39
06:16:47 BLE fromRadio called omlen=1, ourlen=0
06:16:48 [RadioIf] Starting low level send (id=0xb07a448e Fr0xe0 To0xff, WantAck0, HopLim1 encrypted rxtime=1612678607)
06:16:51 [RadioIf] Completed sending (id=0xb07a448e Fr0xe0 To0xff, WantAck0, HopLim1 encrypted rxtime=1612678607)
```

mofosyne, 2021-02-07:
[image]

This is the board I was using

mofosyne, 2021-02-07:
hmmm on last log i noticed

```
00:00:01 RF95 init result -2
00:00:01 Warning: Failed to find RF95 radio
```

Haven't yet booted up a second node to confirm if the modem is not working. But that log messaged doesn't look good. But at least it's not rebooting?

geeksville, 2021-02-08:
@mofosyne  oh yes that second log file looks correct for your rev 1.1 tbeam.  I think the main issue (though I've fixed the rebooting) was that initially you were using a tbeam0.7 load.  I'm glad it worked out.

re 00:00:01 RF95 init result -2
00:00:01 Warning: Failed to find RF95 radio

That is fine.  We try first to find the older RF95 based radios, then if not found (as expected for your sx1262 board) we switch to the better sx1262 driver.