meshtastic/firmware #682: Expose radio details of the current channel for more prominent display in the python API, app, and web server

thebentern, 2021-02-09:
It would be very nice to have the channel details that we only see currently in the logs (such as the operational frequency) exposed by the device in a way that is easily consumed by the python API, app, and web server.
[image]

I have entered a complimentary ticket in the python project to display these attributes in the --info output:
https://github.com/meshtastic/Meshtastic-python/issues/53

geeksville, 2021-02-09:
This issue has been mentioned on **Meshtastic**. There might be relevant details there:

https://meshtastic.discourse.group/t/antenna-improved-range/227/107

mc-hamster, 2021-02-10:
[image]

closed 2021-02-13 by mc-hamster