meshtastic/firmware #685: Incorrectly set enums in protobuf messages

sachaw, 2021-02-12:
I originally discovered an occasional type mismatch in received protobufs when working on the meshtastic.js library, and found that enums are sent as there property name as against there index, i.e. `"modemConfig": "Bw125Cr48Sf4096"` vs `"modemConfig": 3`
This has two issues, firstly messages are much larger than they need to be, secondly, some protobuf libraries aren't smart enough to automatically handle these inconsistencies.

[image]

geeksville, 2021-02-14:
Ooh yes.  Those should definitely be serialized as ints, not strs.  Good find.

geeksville, 2021-02-14:
hmm - where did this json come from?  the web server inside the device or the python api?

sachaw, 2021-02-14:
@geeksville Seems to be an issue in protobufjs which is used by both protoman and meshtastic.js