meshtastic/firmware #688: Map doesn't show any nodes except mine

mervmcrough, 2021-02-14:
**Describe the bug**
Map on the Android App only shows my node and no others. I have cleared cache, data and force stopped the app then reconnected my node but still no dice. i also tried making sure both location and storage permissions were allowed and checked the analytics box (obviouly). I've looked all around the world and zoomed in to specific big cities but it doesn't show a single node except mine anywhere in the world.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Android App'
2. Click on 'Map.'
3. Scroll down to try to find nodes anywhere in the world
4. No error but I know I'm not the only node in the world so where are the others on the map? It's no fun to be the only guy with a node. I know there has to be others. They just aren't showing on the map in the Android app

**Expected behavior**
I expected to see some other nodes on the map

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Device info:**
 - Device model: [e.g. TBEAM] T-Beam 1.1
 - Software Version [e.g. 0.7.8]  1.1.33

**Smartphone information (if relevant):**
 - Device: [e.g. iPhone6] LG Premier Pro
 - OS: [e.g. iOS8.1] Android 8
 - App Version [e.g. 0.7.2] will add later

**Additional context**
Add any other context about the problem here.

geeksville, 2021-02-14:
hmm - can you try updating your devices?  The app contains an update to the 1.1.46 device  code.  It should take about 6 minutes for the app to update the device (you can do this from the device settings panel inside the app).

(1.1.33 should have worked for this - but just to simplify trying to figure out why it isn't working for you)

mervmcrough, 2021-02-15:
Thanks for the quick response. I joined the beta program and updated the firmware so now my app & device are running 1.1.48. Tried uninstalling then reinstalling the app, restarting the T-Beam but no dice. Even if I zoom in on the bay area, no nodes are shown and there's gotta be some there. My node isn't connected to Internet and I havent found a way to do that yet but it seems irrelevant being that my phone running the app is internet connected.
[image]

geeksville, 2021-02-15:
hmm - how many nodes do you (personally) have running.  Does it see them?

(btw: normally I'm in Belmont, CA but in Taiwan for the next few months.  @mc-hamster (the other main dev) is in SSF ish area)

mervmcrough, 2021-02-15:
I only have one (2nd is on the way) and it shows fine. Someone has created a self-reporting map online and that map shows some in the other major cities in USA. I've checked there, western Europe and Asia but can't find another node. Maybe I'll start from scratch again and see if I can make it work. Worst case, when I get the 2nd, it will enable me to do some further troubleshooting.

mervmcrough, 2021-02-15:
By the way, nice work! I love the idea and implementation and I'm happy to be an early adopter

geeksville, 2021-02-16:
Ok - this sounds like expected behavior then.  The [current](https://www.meshtastic.org/) meshtastic (though some folks are experimenting with building bigger meshes, and soon the internet gateway feature might be 'bridging' meshes) is targeted at smallish groups of friends using radios for longish distance remote communication.   The stock antennas can get about 30km with good line of sight, more specialized antennas (usually a cheap moxon) can get a max of about 100km if the line of sight is ideal.

I think even in the bay area the odds of seeing someone elses radio is fairly low.  Though when I'm in town I usually have a moxon pointed from the belmont hills across the bay and it sometimes sees nodes.

you might dig this thread: https://meshtastic.discourse.group/t/mesh-distance-testing-sf-oakland-berkeley/1036

If you have other questions/comments the forum  probably gets more readership than github issues (though issues are fine also).

closed 2021-02-16 by geeksville

Oyshoboy, 2025-07-12:
same issue on iOS [image]

vicnaum, 2025-10-25:
Here it doesn't even show mine on the map :) Altho I enabled the phone GPS (my Haltec V3 doesn't have it's own GPS) and set a fixed position, etc. Still didn't appear on the map and doesn't have a location when I try to "Exchange Location"