meshtastic/firmware #691: SPIFFS Enumerating Weirdly

crossan007, 2021-02-15:
Running @mc-hamster's branch at `8320754b985e8c8cebb42d72c3287fb1dab45d74`.

I'm seeing some weird duplication of files on SIPFFS:
[image]

After a full `factory_reset`, I'm seeing "shadows" of files that were previously on the device, but with 0 bytes:
[image]

crossan007, 2021-02-15:
Trying to delete one of these 0-byte static files shows this in `meshtastic --debug` logs:

```
[HTTPS:I] Request: GET /static?delete=static/index.html.gz (FID=59)
[E][vfs_api.cpp:238] VFSFileImpl(): fopen(/spiffs/static/index.html.gz) failed
[E][vfs_api.cpp:135] remove(): /static/index.html.gz does not exists or is directory
/static/index.html.gz
Trigger powerFSM 9
[HTTPS:I] Connection closed. Socket FID=59
[HTTPS:I] New connection. Socket FID=60
[HTTPS:I] Request: GET /static (FID=60)
```

crossan007, 2021-02-15:
After a hard-reset, the files on SPIFFS now appear with non-zero sizes 🤔🤔🤔

geeksville, 2021-02-16:
hehehe!

mc-hamster, 2021-02-16:
Espressif is moving away from SPIFFS. It wouldn't surprise me that in a not so distant future SDK, SPIFFS would be deprecated entirely. Corruption is a frequent complaint.

mc-hamster, 2021-03-18:
Closing. Please reopen if you see it again.

closed 2021-03-18 by mc-hamster