meshtastic/firmware #696: User pref to disable BLE / Wifi completely (or manually enable it)

thebentern, 2021-02-17:
After doing some experimentation for my use case of long term low power repeater deployments, I believe that bluetooth / wifi is unnecessary for this application. It would be a really nice compliment to the router mode / low power prefs to have one that just disables BLE / Wifi. I will be configuring and performing maintenance via usb serial. 

Another alternative option that might be preferable is to define a setting for GPIO pin mapping for a button on the board to initiate bluetooth manually. I think meshtastic-web already does something similar on a predefined pin for starting up the web service. This could prove very useful for OTA flashes of remote nodes that otherwise have no use for bluetooth.

geeksville, 2021-02-17:
> low power prefs to have one that just disables BLE / Wifi. I will most
> likely be configuring and performing maintenance via usb serial.

There is already an option for this.  I'll write more when I get home.

(Sent from a phone - please ignore typos)

crossan007, 2021-02-18:
there is a mechanism in meshtastic-device which allows you to hold the user-button during device start-up to force the device into a SoftAP mode.  

not a function of meshtastic-web, but certainly useful for accessing meshtastic-web on a previously unconfigured device (https://github.com/meshtastic/Meshtastic-device/releases/tag/1.1.23)

thebentern, 2021-02-18:
> not a function of meshtastic-web, but certainly useful for accessing meshtastic-web on a previously unconfigured device 
Right! That's what I was trying to think of. Sorry for conflating the device and meshtastic-web. I didn't separate the two in my head since I have meshtastic-web going on my home base station device.

mc-hamster, 2021-12-25:
@thebentern This is nearly a year old. Still needed?

thebentern, 2021-12-26:
@mc-hamster, I'm going to close this because I think it's rendered pretty irrelevant with my experience of NRF based devices and their low power consumption. Better to solve the problem with suitable hardware solution rather than bending the software around to make small gains. :-)

closed 2021-12-26 by thebentern