meshtastic/firmware #697: WiFi switches to SoftAP upon reset/restart after setting it in config

zyrorl, 2021-02-18:
Hi there,

So I've setup my two ttgo lora32 devices on the latest 1.1.48 beta, and it appears that when i first set up the wifi to use my local wifi network, everything works fine, however when i restart the devices they turn on SoftAP instead.

I'm using the instructions in the python lib to set this up

**To Reproduce**
```
meshtastic --port /dev/DEVICEID --set wifi_ap_mode false --setstr wifi_ssid myWifiSSID --setstr wifi_password password
```

But the next time the device restarts it goes back to softAP, either by power switch or reset button.

**Expected behaviour**
The device re-connects to the wifi network on reboot/power on.

**Device info:**
- TTGO Lora32 V2 T3_V1.6

**Additional question**
Also is there a way to disable WIFI all together for power saving? I'd like to do this via a button or something.

crossan007, 2021-02-20:
From discussion in Slack:
```
I wonder if that's causing the software to "think" it's being held during boot.   I ask because that will force it into soft ap mode.
Could the floating voltage trigger it?
```

https://meshtasticdev.slack.com/archives/C016JMLCD9Q/p1613772653006200

crossan007, 2021-02-20:
From Slack:
```
Ben Meadors  6:36 PM
I think you're on to something. It may be defaulting to high. I just booted it with 5v jumped to that pin and voila... no AP.
```

mc-hamster, 2021-02-20:
Root cause:

we expect a button on pin 12 but there is no button on the device.

instead, expect the user to wire a button to that pin and ground.

to help the user, we enable the internal pull-up for that pin.

unfortunately, there's a bidirectional diode on that pin that is pulling the line low.

[image]

this causes the device to read that the pin is being held down.

mc-hamster, 2021-02-20:
Final solution is to disable the internal pull-up. GPIO 12 is the only usable pin. It would need an external pull up stronger than what's available on the esp32.

zyrorl, 2021-02-20:
@mc-hamster so i have a current button wired on GPIO12 for the page switching, am i supposed to move it to another GPIO? or what exactly do i need to do?

mc-hamster, 2021-02-20:
Still working on this. My original two solutions were not exactly right.

mc-hamster, 2021-02-20:
@zyrorl How do you have the button attached to GPIO 12? How do you have it wired?

mc-hamster, 2021-02-20:
GPIO 12 is an odd pin. The internal pull up is slow to be enabled. That caused what I mistook as hysteresis. This is because of the slightly longer logic chain on silicon in the the MUX to support SD cards.

Fix is to add a slight delay when enabling the GPIO pull up to allow the logic gates to settle to the proper state.

closed 2021-02-23 by mc-hamster