meshtastic/firmware #700: TBeam Periodically Rebooting

crossan007, 2021-02-20:
On the latest commit of `master` (f6e2695a4d095020a7962e1048b1d29fe9480e5e)

I'm getting random reboots on my TBeams:
```
PS C:\Users\Charles\Documents\GitHub\Meshtastic\Meshtastic-device\bin> python .\exception_decoder.py -e ..\.pio\build\tbeam\firmware.elf -f .\backtrace.txt
stack:
0:
  0x4008ff5f: invoke_abort at /home/kevinh/development/meshtastic/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:715
0:
  0x400902e5: abort at /home/kevinh/development/meshtastic/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:715
0:
  0x4010f39f: __assert_func at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdlib/../../../.././newlib/libc/stdlib/assert.c:63 (discriminator 8)
0:
  0x400dd527: Allocator<_MeshPacket>::allocZeroed() at C:\Users\Charles\Documents\GitHub\Meshtastic\Meshtastic-device/src/mesh/RadioLibInterface.cpp:319
  \-> inlined by: RadioLibInterface::handleReceiveInterrupt() at C:\Users\Charles\Documents\GitHub\Meshtastic\Meshtastic-device/src/mesh/RadioLibInterface.cpp:272
0:
  0x400dd553: RadioLibInterface::onNotify(unsigned int) at C:\Users\Charles\Documents\GitHub\Meshtastic\Meshtastic-device/src/mesh/RadioLibInterface.cpp:319
0:
  0x400dd5d4: non-virtual thunk to RadioLibInterface::onNotify(unsigned int) at ??:?
0:
  0x401d9587: concurrency::NotifiedWorkerThread::runOnce() at C:\Users\Charles\Documents\GitHub\Meshtastic\Meshtastic-device/src/concurrency/NotifiedWorkerThread.cpp:79
0:
  0x400d4d86: concurrency::OSThread::run() at C:\Users\Charles\Documents\GitHub\Meshtastic\Meshtastic-device/src/concurrency/OSThread.cpp:45
0:
  0x400eef35: ThreadController::runOrDelay() at C:\Users\Charles\Documents\GitHub\Meshtastic\Meshtastic-device/.pio/libdeps/tbeam/Thread/ThreadController.cpp:153
0:
  0x400d9db8: loop() at C:\Users\Charles\Documents\GitHub\Meshtastic\Meshtastic-device/src/main.cpp:609
0:
  0x400fe1c9: loopTask(void*) at C:/Users/Charles/.platformio/packages/framework-arduinoespressif32@src-919133ccb22f3315e5406013d6432bdf/cores/esp32/main.cpp:19
  ```

crossan007, 2021-02-20:
For context, the device is connected to WiFi.

"New" settings that are enabled are:
`rangeTestPluginEnabled` is true
`rangeTestPluginSave` is true

crossan007, 2021-02-20:
another stack trace:
```
                                                                       python .\exception_decoder.py -e ..\.pio\build\tbeam\firmware.elf -f .\backtrace2.txt
stack:
0:
  0x400812b4: RadioLibInterface::isrLevel0Common(RadioLibInterface::PendingISR) at C:\Users\Charles\Documents\GitHub\Meshtastic\Meshtastic-device/src/mesh/RadioLibInterface.cpp:319
0:
  0x400812dd: RadioLibInterface::isrRxLevel0() at C:\Users\Charles\Documents\GitHub\Meshtastic\Meshtastic-device/src/mesh/RadioLibInterface.cpp:319
0:
  0x40081359: __onPinInterrupt at C:/Users/Charles/.platformio/packages/framework-arduinoespressif32@src-919133ccb22f3315e5406013d6432bdf/cores/esp32/esp32-hal-gpio.c:274
0:
  0x4008246d: _xt_lowint1 at /home/kevinh/development/meshtastic/esp32-arduino-lib-builder/esp-idf/components/freertos/xtensa_vectors.S:1154
0:
  [DATA (0x0x40062232)]
0:
  0x40096879: esp_rom_spiflash_read_status at /home/kevinh/development/meshtastic/esp32-arduino-lib-builder/esp-idf/components/spi_flash/spi_flash_rom_patch.c:413
0:
  0x400968b2: esp_rom_spiflash_wait_idle at /home/kevinh/development/meshtastic/esp32-arduino-lib-builder/esp-idf/components/spi_flash/spi_flash_rom_patch.c:413
0:
  0x40096c09: esp_rom_spiflash_erase_sector_internal at /home/kevinh/development/meshtastic/esp32-arduino-lib-builder/esp-idf/components/spi_flash/spi_flash_rom_patch.c:121        
  \-> inlined by: esp_rom_spiflash_erase_sector at /home/kevinh/development/meshtastic/esp32-arduino-lib-builder/esp-idf/components/spi_flash/spi_flash_rom_patch.c:451
0:
  0x40086c15: spi_flash_erase_range at /home/kevinh/development/meshtastic/esp32-arduino-lib-builder/esp-idf/components/spi_flash/flash_ops.c:237
0:
  0x4017e55d: esp_partition_erase_range at /home/kevinh/development/meshtastic/esp32-arduino-lib-builder/esp-idf/components/spi_flash/partition.c:294
0:
  0x40186f15: spiffs_api_erase at /home/kevinh/development/meshtastic/esp32-arduino-lib-builder/esp-idf/components/spiffs/spiffs_api.c:58
0:
  0x401886e8: spiffs_erase_block at /home/kevinh/development/meshtastic/esp32-arduino-lib-builder/esp-idf/components/spiffs/spiffs/src/spiffs_nucleus.c:1164
0:
  0x4018accd: spiffs_gc_erase_block at /home/kevinh/development/meshtastic/esp32-arduino-lib-builder/esp-idf/components/spiffs/spiffs/src/spiffs_gc.c:15
0:
  0x4018b6c6: spiffs_gc_check at /home/kevinh/development/meshtastic/esp32-arduino-lib-builder/esp-idf/components/spiffs/spiffs/src/spiffs_gc.c:168
0:
  0x40189365: spiffs_object_append at /home/kevinh/development/meshtastic/esp32-arduino-lib-builder/esp-idf/components/spiffs/spiffs/src/spiffs_nucleus.c:1219
0:
  0x401871f2: spiffs_hydro_write at /home/kevinh/development/meshtastic/esp32-arduino-lib-builder/esp-idf/components/spiffs/spiffs/src/spiffs_hydrogen.c:1121
0:
  0x40187257: spiffs_fflush_cache at /home/kevinh/development/meshtastic/esp32-arduino-lib-builder/esp-idf/components/spiffs/spiffs/src/spiffs_hydrogen.c:1121
0:
  0x40187b74: SPIFFS_close at /home/kevinh/development/meshtastic/esp32-arduino-lib-builder/esp-idf/components/spiffs/spiffs/src/spiffs_hydrogen.c:1121
0:
  0x40186a69: vfs_spiffs_close at /home/kevinh/development/meshtastic/esp32-arduino-lib-builder/esp-idf/components/spiffs/esp_spiffs.c:496
0:
  0x40191e41: esp_vfs_close at /home/kevinh/development/meshtastic/esp32-arduino-lib-builder/esp-idf/components/vfs/vfs.c:734
0:
  [DATA (0x0x4000bd4b)]
0:
  [DATA (0x0x400011c1)]
0:
  [DATA (0x0x40002055)]
0:
  [DATA (0x0x400020b5)]
0:
  0x401c4c0b: VFSFileImpl::close() at C:/Users/Charles/.platformio/packages/framework-arduinoespressif32@src-919133ccb22f3315e5406013d6432bdf/libraries/FS/src/vfs_api.h:31
0:
  0x400f4a53: fs::File::close() at C:/Users/Charles/.platformio/packages/framework-arduinoespressif32@src-919133ccb22f3315e5406013d6432bdf/libraries/FS/src/FS.cpp:258
0:
  0x400e69a6: RangeTestPluginRadio::appendFile(_MeshPacket const&) at C:\Users\Charles\Documents\GitHub\Meshtastic\Meshtastic-device/src/plugins/esp32/RangeTestPlugin.cpp:312
0:
  0x400e6aa1: RangeTestPluginRadio::handleReceived(_MeshPacket const&) at C:\Users\Charles\Documents\GitHub\Meshtastic\Meshtastic-device/src/plugins/esp32/RangeTestPlugin.cpp:150
0:
  0x400dad6c: MeshPlugin::callPlugins(_MeshPacket const&) at C:\Users\Charles\Documents\GitHub\Meshtastic\Meshtastic-device/src/mesh/MeshPlugin.cpp:38
0:
  0x400dbb56: NodeDB::updateFrom(_MeshPacket const&) at c:\users\charles\.platformio\packages\toolchain-xtensa32\xtensa-esp32-elf\include\c++\5.2.0\bits/shared_ptr_base.h:167
0:
  0x400db150: MeshService::handleFromRadio(_MeshPacket const*) at C:\Users\Charles\Documents\GitHub\Meshtastic\Meshtastic-device/src/mesh/MeshService.cpp:77
0:
  0x401d9756: CallbackObserver<MeshService, _MeshPacket const*>::onNotify(_MeshPacket const*) at C:\Users\Charles\Documents\GitHub\Meshtastic\Meshtastic-device/src/Observer.h:50 (discriminator 4)
0:
  0x400de089: Observable<_MeshPacket const*>::notifyObservers(_MeshPacket const*) at C:\Users\Charles\Documents\GitHub\Meshtastic\Meshtastic-device/src/Observer.h:71
  \-> inlined by: Router::handleReceived(_MeshPacket*) at C:\Users\Charles\Documents\GitHub\Meshtastic\Meshtastic-device/src/mesh/Router.cpp:270
0:
  0x400de1ea: Router::perhapsHandleReceived(_MeshPacket*) at C:\Users\Charles\Documents\GitHub\Meshtastic\Meshtastic-device/src/mesh/Router.cpp:289
0:
  0x400de211: Router::runOnce() at C:\Users\Charles\Documents\GitHub\Meshtastic\Meshtastic-device/src/mesh/Router.cpp:56
0:
[... cut ...]
  0x400d4d86: concurrency::OSThread::run() at C:\Users\Charles\Documents\GitHub\Meshtastic\Meshtastic-device/src/concurrency/OSThread.cpp:45
0:
  0x400eef35: ThreadController::runOrDelay() at C:\Users\Charles\Documents\GitHub\Meshtastic\Meshtastic-device/.pio/libdeps/tbeam/Thread/ThreadController.cpp:153
0:
  0x400d9db8: loop() at C:\Users\Charles\Documents\GitHub\Meshtastic\Meshtastic-device/src/main.cpp:609
0:
  0x400fe1c9: loopTask(void*) at C:/Users/Charles/.platformio/packages/framework-arduinoespressif32@src-919133ccb22f3315e5406013d6432bdf/cores/esp32/main.cpp:19
  ```

crossan007, 2021-02-20:
And another crash:

```
PS C:\Users\Charles\Documents\GitHub\Meshtastic\Meshtastic-device\bin> python .\exception_decoder.py -e ..\.pio\build\tbeam\firmware.elf -f .\backtrace3.txt
stack:
0:
  0x400812b4: RadioLibInterface::isrLevel0Common(RadioLibInterface::PendingISR) at C:\Users\Charles\Documents\GitHub\Meshtastic\Meshtastic-device/src/mesh/RadioLibInterface.cpp:319
0:
  0x400812dd: RadioLibInterface::isrRxLevel0() at C:\Users\Charles\Documents\GitHub\Meshtastic\Meshtastic-device/src/mesh/RadioLibInterface.cpp:319
0:
  0x40081359: __onPinInterrupt at C:/Users/Charles/.platformio/packages/framework-arduinoespressif32@src-919133ccb22f3315e5406013d6432bdf/cores/esp32/esp32-hal-gpio.c:274
0:
  0x4008246d: _xt_lowint1 at /home/kevinh/development/meshtastic/esp32-arduino-lib-builder/esp-idf/components/freertos/xtensa_vectors.S:1154
0:
  0x40086421: spi_flash_op_block_func at /home/kevinh/development/meshtastic/esp32-arduino-lib-builder/esp-idf/components/spi_flash/cache_utils.c:201
0:
  0x40083731: ipc_task at /home/kevinh/development/meshtastic/esp32-arduino-lib-builder/esp-idf/components/esp32/ipc.c:62```

crossan007, 2021-02-20:
Another backtrace using build from 621313d 

```
PS C:\Users\Charles\Documents\GitHub\Meshtastic\Meshtastic-device\bin> python .\exception_decoder.py -e ..\.pio\build\tbeam\firmware.elf -f ..\Backtraces\backtrace.txt
stack:
0:
  0x400812b4: RadioLibInterface::isrLevel0Common(RadioLibInterface::PendingISR) at C:\Users\Charles\Documents\GitHub\Meshtastic\Meshtastic-device/src/mesh/RadioLibInterface.cpp:319
0:
  0x400812dd: RadioLibInterface::isrRxLevel0() at C:\Users\Charles\Documents\GitHub\Meshtastic\Meshtastic-device/src/mesh/RadioLibInterface.cpp:319
0:
  0x40081359: __onPinInterrupt at C:/Users/Charles/.platformio/packages/framework-arduinoespressif32@src-919133ccb22f3315e5406013d6432bdf/cores/esp32/esp32-hal-gpio.c:274
0:
  0x4008246d: _xt_lowint1 at /home/kevinh/development/meshtastic/esp32-arduino-lib-builder/esp-idf/components/freertos/xtensa_vectors.S:1154
0:
  [DATA (0x0x40062235)]
0:
  0x40096879: esp_rom_spiflash_read_status at /home/kevinh/development/meshtastic/esp32-arduino-lib-builder/esp-idf/components/spi_flash/spi_flash_rom_patch.c:413
0:
  0x400968b2: esp_rom_spiflash_wait_idle at /home/kevinh/development/meshtastic/esp32-arduino-lib-builder/esp-idf/components/spi_flash/spi_flash_rom_patch.c:413
0:
  0x400969dd: esp_rom_spiflash_program_page_internal at /home/kevinh/development/meshtastic/esp32-arduino-lib-builder/esp-idf/components/spi_flash/spi_flash_rom_patch.c:413
0:
  0x40096c7e: esp_rom_spiflash_write at /home/kevinh/development/meshtastic/esp32-arduino-lib-builder/esp-idf/components/spi_flash/spi_flash_rom_patch.c:481
0:
  0x40086d92: spi_flash_write_inner at /home/kevinh/development/meshtastic/esp32-arduino-lib-builder/esp-idf/components/spi_flash/flash_ops.c:271
  \-> inlined by: spi_flash_write at /home/kevinh/development/meshtastic/esp32-arduino-lib-builder/esp-idf/components/spi_flash/flash_ops.c:402
0:
  0x4017e537: esp_partition_write at /home/kevinh/development/meshtastic/esp32-arduino-lib-builder/esp-idf/components/spi_flash/partition.c:274
0:
  0x40186efa: spiffs_api_write at /home/kevinh/development/meshtastic/esp32-arduino-lib-builder/esp-idf/components/spiffs/spiffs_api.c:47
0:
  0x4018abd5: spiffs_phys_wr at /home/kevinh/development/meshtastic/esp32-arduino-lib-builder/esp-idf/components/spiffs/spiffs/src/spiffs_cache.c:221
0:
  0x401884c9: spiffs_phys_cpy at /home/kevinh/development/meshtastic/esp32-arduino-lib-builder/esp-idf/components/spiffs/spiffs/src/spiffs_nucleus.c:1164
0:
  0x40188d8d: spiffs_page_move at /home/kevinh/development/meshtastic/esp32-arduino-lib-builder/esp-idf/components/spiffs/spiffs/src/spiffs_nucleus.c:1164
0:
  0x4018b2b9: spiffs_gc_clean at /home/kevinh/development/meshtastic/esp32-arduino-lib-builder/esp-idf/components/spiffs/spiffs/src/spiffs_gc.c:448
0:
  0x4018b6c9: spiffs_gc_check at /home/kevinh/development/meshtastic/esp32-arduino-lib-builder/esp-idf/components/spiffs/spiffs/src/spiffs_gc.c:156
0:
  0x40189385: spiffs_object_append at /home/kevinh/development/meshtastic/esp32-arduino-lib-builder/esp-idf/components/spiffs/spiffs/src/spiffs_nucleus.c:1219
0:
  0x40187212: spiffs_hydro_write at /home/kevinh/development/meshtastic/esp32-arduino-lib-builder/esp-idf/components/spiffs/spiffs/src/spiffs_hydrogen.c:1121
0:
  0x40187277: spiffs_fflush_cache at /home/kevinh/development/meshtastic/esp32-arduino-lib-builder/esp-idf/components/spiffs/spiffs/src/spiffs_hydrogen.c:1121
0:
  0x40187b94: SPIFFS_close at /home/kevinh/development/meshtastic/esp32-arduino-lib-builder/esp-idf/components/spiffs/spiffs/src/spiffs_hydrogen.c:1121
0:
  0x40186a89: vfs_spiffs_close at /home/kevinh/development/meshtastic/esp32-arduino-lib-builder/esp-idf/components/spiffs/esp_spiffs.c:496
0:
  0x40191e61: esp_vfs_close at /home/kevinh/development/meshtastic/esp32-arduino-lib-builder/esp-idf/components/vfs/vfs.c:734
0:
  [DATA (0x0x4000bd4b)]
0:
  [DATA (0x0x400011c1)]
0:
  [DATA (0x0x40002055)]
0:
  [DATA (0x0x400020b5)]
0:
  0x401c4c2b: VFSFileImpl::close() at C:/Users/Charles/.platformio/packages/framework-arduinoespressif32@src-919133ccb22f3315e5406013d6432bdf/libraries/FS/src/vfs_api.h:31
0:
  0x400f4a73: fs::File::close() at C:/Users/Charles/.platformio/packages/framework-arduinoespressif32@src-919133ccb22f3315e5406013d6432bdf/libraries/FS/src/FS.cpp:258
0:
  0x400e69c6: RangeTestPluginRadio::appendFile(_MeshPacket const&) at C:\Users\Charles\Documents\GitHub\Meshtastic\Meshtastic-device/src/plugins/esp32/RangeTestPlugin.cpp:312
0:
  0x400e6ac1: RangeTestPluginRadio::handleReceived(_MeshPacket const&) at C:\Users\Charles\Documents\GitHub\Meshtastic\Meshtastic-device/src/plugins/esp32/RangeTestPlugin.cpp:150
0:
  0x400dad8c: MeshPlugin::callPlugins(_MeshPacket const&) at C:\Users\Charles\Documents\GitHub\Meshtastic\Meshtastic-device/src/mesh/MeshPlugin.cpp:38
0:
  0x400dbb76: NodeDB::updateFrom(_MeshPacket const&) at c:\users\charles\.platformio\packages\toolchain-xtensa32\xtensa-esp32-elf\include\c++\5.2.0\bits/shared_ptr_base.h:167
0:
  0x400db170: MeshService::handleFromRadio(_MeshPacket const*) at C:\Users\Charles\Documents\GitHub\Meshtastic\Meshtastic-device/src/mesh/MeshService.cpp:77
0:
  0x401d9776: CallbackObserver<MeshService, _MeshPacket const*>::onNotify(_MeshPacket const*) at C:\Users\Charles\Documents\GitHub\Meshtastic\Meshtastic-device/src/Observer.h:50 (discriminator 4)
0:
[... cut ...]
  0x400d4da6: concurrency::OSThread::run() at C:\Users\Charles\Documents\GitHub\Meshtastic\Meshtastic-device/src/concurrency/OSThread.cpp:45
0:
  0x400eef55: ThreadController::runOrDelay() at C:\Users\Charles\Documents\GitHub\Meshtastic\Meshtastic-device/.pio/libdeps/tbeam/Thread/ThreadController.cpp:153
0:
  0x400d9dd8: loop() at C:\Users\Charles\Documents\GitHub\Meshtastic\Meshtastic-device/src/main.cpp:609
0:
  0x400fe1e9: loopTask(void*) at C:/Users/Charles/.platformio/packages/framework-arduinoespressif32@src-919133ccb22f3315e5406013d6432bdf/cores/esp32/main.cpp:19
  ```

mc-hamster, 2021-03-28:
Closing this. No one else has reported this condition. I suspect a corrupt spiffs.

closed 2021-03-28 by mc-hamster