meshtastic/firmware #704: GPIO 13 doesnt work for buzzer on TT GO lora boards

zyrorl, 2021-02-21:
latest code in master activates buzzer on gpio 13 for TTGO Lora T3 V2.1.6 even if there's no gps attached

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Device info:**
 - Device model: [e.g. TBEAM]
 - Software Version [e.g. 0.7.8]

**Smartphone information (if relevant):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - App Version [e.g. 0.7.2]

**Additional context**
Add any other context about the problem here.

mc-hamster, 2021-02-21:
There are three causes of this.

1) A recent change sets the GPS TX pin for that board to GPIO 13.
2) The code to detect that the GPS is initialized doesn't look in the right place.
3) _serial_gps->end(); is never run to stop sending out of the GPS TX if no gps is found.

garthvh, 2022-08-04:
https://github.com/meshtastic/Meshtastic-device/pull/1583

closed 2022-08-04 by garthvh