meshtastic/firmware #707: "00:01:08 68 [EnvironmentalMeasurementPlugin] Environmental Measurement Plugin: DISABLED; The SENSOR_READ_ERROR_COUNT_THRESHOLD has been exceed: 5"

geeksville, 2021-02-22:
I see this repeatedly in the logs on my 1.2 dev branch.  Is this intended?  Did I mess things up wrt master?

Not urgent but fyi

crossan007, 2021-02-22:
Yeah...   I have a mr coming soon to fix this.

It's because the plug-in is enabled by default, and you likely don't have a dht11 connected.

crossan007, 2021-02-22:
It tries to read 5 times, and then gives up

crossan007, 2021-02-26:
This should be resolved with #705.

The behavior is now documented in the protobufs: https://github.com/meshtastic/Meshtastic-protobufs/blob/master/mesh.proto#L929

sachaw, 2021-03-27:
@geeksville is this solved?
@crossan007 any chance you can abbreviate your plugin naming, `environmentalMeasurementPluginMeasurementEnabled` is unsightly, maybe `envSensorPluginEnabled` or something along those lines.

garthvh, 2022-01-01:
Is this still and open issue?

crossan007, 2022-01-01:
We can probably close this?   

To be honest, I haven't really played with Meshtastic since fixing this issue.

thebentern, 2022-01-01:
Closing. Feel free to open a new one if you get back into it.

closed 2022-01-01 by thebentern