meshtastic/firmware #714: Bluetooth not disabling when usb connected operating over API

ohcdh, 2021-02-25:
https://meshtastic.discourse.group/t/disable-bluetooth/2640

Please see video showing blue led for bluetooth flashing when usb cable connected.  I have also enabled bluetooth on the phone after i have connected and the t-beam is showing up as a device that can be paired.

http://www.fileconvoy.com/dfl.php?id=g7dbd8f0b000b641f10003311285e6e0bc5d8c879a9****

geeksville, 2021-03-11:
This issue has been mentioned on **Meshtastic**. There might be relevant details there:

https://meshtastic.discourse.group/t/disable-bluetooth/2640/13

ohcdh, 2021-03-11:
Thanks @geeksville I opened the thread on discourse, I have upgraded the device to 1.2.6 but when device is connected to phone, bluetooth is still active on device

ohcdh, 2021-03-20:
it turns out that when a personal radio is powered by a battery and connected by usb cable the bluetooth de-activates, so this is perfect.  It was my misunderstanding, as I was testing devices without a battery and powered solely by usb and therefore bluetooth was activated. I didn't realise that the radio differentiated between usb power and usb api.

closed 2021-03-20 by ohcdh