meshtastic/firmware #731: Packet source tracking

iz1kga, 2021-03-05:
Hello everyone,

This bug is for ask a new feature that allows to track packet source when it is repeated.

Some details on discourse:
https://meshtastic.discourse.group/t/packet-source-tracking/2764/2

Use case:
Allow the routing tracking capability adding data to an array containing routing history for eg. to draw the mesh configuration on a map or to understand the best position of a repeater node.

User Needs:
- User needs a variable containing routing history
- User needs the possibility to enable or disable the feature when this is not needed for the application

MrSniffer, 2021-03-16:
It would be helpful if the routing information also included the rssi for each hop.

garthvh, 2022-01-27:
There is quite a bit of logging of mesh traffic on both mobile apps and via serial and the CLI.

closed 2022-01-27 by garthvh