meshtastic/firmware #753: Missing pins_arduino.h when building tbeam

timgunter, 2021-03-22:
I submitted a pull request for a minor change to the device install/update shell scripts. I noticed that this PR failed one of the CI tests that builds the tbeam firmware: [https://github.com/meshtastic/Meshtastic-device/pull/751/checks?check_run_id=2162344812](url)

PR: [https://github.com/meshtastic/Meshtastic-device/pull/751](url)

As I noted in the PR, I have seen something similar when building the firmware on a raspberry pi; tbeam reports "pins_arduino.h" missing, and heltec build succeeds.

meehow, 2021-03-26:
I think https://github.com/meshtastic/arduino-esp32/pull/3 should fix that

mc-hamster, 2021-03-27:
Closing this bug. Apologize you ran into it.

The CI test are currently being reworked to build support for running the code in a simulator and executing unit tests.

Once that effort is done the CI will be back up and running. For now we're manually reviewing PRs with detail.

closed 2021-03-27 by mc-hamster